introlab/rtabmap #1207: Errors compiling from source on Windows 11

DenTechs, 2024-01-26:
I've spent the better part of a day trying to figure out where i'm going wrong, but as a last resort i'll try to ask here. I've followed the instructions in the wiki for building rtab-map from source with vcpkg for dependencies and cmake/visual studio to compile, however i'm running into issues i dont understand. Below are the log files from visual studio:

Severity	Code	Description	Project	File	Line	Suppression State	Details
Warning	C4267	'argument': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\Compression.cpp	280		
Error	C2121	'#': invalid character: possibly the result of a macro expansion	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\camera\CameraStereoZed.cpp	529		
Error	C2059	syntax error: 'if'	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\camera\CameraStereoZed.cpp	517		
Error	C2121	'#': invalid character: possibly the result of a macro expansion	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\camera\CameraStereoZed.cpp	545		
Error	C2059	syntax error: 'if'	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\camera\CameraStereoZed.cpp	542		
Error	C1019	unexpected #else	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\camera\CameraStereoZed.cpp	557		
Warning	C4267	'+=': conversion from 'size_t' to 'unsigned long', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\VisualWord.cpp	75		
Warning	C4267	'+=': conversion from 'size_t' to 'unsigned long', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\VisualWord.cpp	76		
Warning	C4267	'+=': conversion from 'size_t' to 'unsigned long', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\VisualWord.cpp	77		
Warning	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\util3d.cpp	3374		
Warning	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\util\serialization.h	484		
Warning	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\util\serialization.h	751		
Warning	C4996	'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\util\saving.h	58		
Warning	C4996	'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\util\saving.h	60		
Warning	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\util\logger.h	66		
Warning	C4996	'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\flann.hpp	383		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\hierarchical_clustering_index.h	318		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\hierarchical_clustering_index.h	535		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	1054		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	1056		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	729		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\hierarchical_clustering_index.h	445		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	704		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\hierarchical_clustering_index.h	252		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	251		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\hierarchical_clustering_index.h	413		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	137		
Warning	C4291	'void *operator new(size_t,rtflann::PooledAllocator &)': no matching operator delete found; memory will not be freed if initialization throws an exception	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\rtflann\algorithms\kdtree_index.h	138		
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	605		
Warning	C4267	'=': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	686		
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	692		
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	847		
Warning	C4267	'initializing': conversion from 'size_t' to 'const int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	847		
Warning	C4267	'initializing': conversion from 'size_t' to 'int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	952		
Warning	C4267	'initializing': conversion from 'size_t' to 'const int', possible loss of data	rtabmap_core	C:\Users\dentechs\Documents\GitHub\rtabmap\corelib\src\opencv\ORBextractor.cc	952		
[... cut ...]
Error	LNK1181	cannot open input file '..\..\lib\rtabmap_gui.lib'	wifi_mapping	C:\Users\dentechs\Documents\GitHub\rtabmap\build\examples\WifiMapping\LINK	1		
Error	LNK1181	cannot open input file '..\..\lib\rtabmap_gui.lib'	noEventsExample	C:\Users\dentechs\Documents\GitHub\rtabmap\build\examples\NoEventsExample\LINK	1		
Error	LNK1181	cannot open input file '..\..\lib\rtabmap_gui.lib'	matcher	C:\Users\dentechs\Documents\GitHub\rtabmap\build\tools\Matcher\LINK	1		
Error	LNK1181	cannot open input file '..\..\lib\rtabmap_gui.lib'	rgbd_mapping	C:\Users\dentechs\Documents\GitHub\rtabmap\build\examples\RGBDMapping\LINK	1

matlabbe, 2024-01-28:
Is it during the step `vcpkg install rtabmap --clean-after-build` ? https://github.com/introlab/rtabmap/wiki/Installation#vcpkg

DenTechs, 2024-01-28:
> Is it during the step `vcpkg install rtabmap --clean-after-build` ? https://github.com/introlab/rtabmap/wiki/Installation#vcpkg

No, as far as i can tell the vcpkg version of rtabmap is the same as the current release on the github, which is on an outdated version of the Zed sdk. I saw that the master branch of rtabmap was already updated to it however, so i've been trying to compile it manually with cmake. Those errors are occurring when building with cmake or in visual studio 2022.

matlabbe, 2024-01-28:
I compiled it last weekend on windows, though I didn't re-update all vcpkg dependencies. I'll try updating everything and see if I can reproduce your error.

DenTechs, 2024-01-28:
Thanks! I appreciate the help

matlabbe, 2024-01-31:
The ZED errors have been fixed in this commit: https://github.com/introlab/rtabmap/commit/b2a86d640a15ebaf0b189b3721c978d8645018d7

Unless you use the opencv version included with ZED SDK, **remove or comment** the following line inside `zed-config.cmake` to avoid many opencv related link errors:
```cmake
SET(OpenCV_DIR ${ZED_DIR}/dependencies/opencv_3.1.0 CACHE BOOL "" FORCE)
```

cheers,
Mathieu