introlab/rtabmap #1208: Generate obj,  mtl, and texture jpg incrementally

360wcui, 2024-01-26:
Hello,  

Can we incrementally render the mesh model (obj, mtl, and jpg) of the 3D map instead of the point cloud from mapData or other sources?  I like the way how the iPhone version does it. 

Any suggestions would be greatly appreciated.    Thank you.

matlabbe, 2024-01-28:
With standalone or rtabmap_viz in ros, it is possible to show meshes in real-time like the iPhone app does. Open Preferences and check this box:
[image]

This only works for RGB-D data. You may increase the decimation to avoid adding too many polygons.

In ROS, outside rtabmap_viz, there is no way I am aware to do that with data from MapData topic.

360wcui, 2024-01-28:
Thank you!  Are the mesh models (e.g., obj, mtl) saved anywhere for rtabmap_viz to consume or which code snippet does online meshing?

matlabbe, 2024-01-30:
The meshes are kept in renderer memory. To export the mesh, you would use File->Export Clouds... and check Meshing box.

The code creating the meshes online is here:
https://github.com/introlab/rtabmap/blob/master/guilib/src/MainWindow.cpp#L3796-L3866

closed 2024-01-31 by 360wcui