introlab/rtabmap #1209: Which depth? Distance to plane or to optical center?

MotorTrick, 2024-01-26:
Hi there, 
I'm wondering if we should provide the depth as distance between the camera position and the points in the world (i.e. raycast from optical center to object), or the depth as distance from the image plane (so if I'm pointing the camera to a flat wall everything will have the same reading).

Is there somewhere in the code in which you take care of this?

matlabbe, 2024-01-28:
See https://ros.org/reps/rep-0118.html:

> Each pixel is a depth (along the camera Z axis) in meters.

That would match more to the "distance to plane" you are describing. Intuitively, if the camera is looking straight to a flat wall, all depth values will be the same like you said.

MotorTrick, 2024-01-29:
Thanks. Do you know a way through which I can convert between the two? :pray:

matlabbe, 2024-01-30:
You could project each pixel of your depth image in 3D, then the `z` value of the 3d point will be your depth.

You may use this function [image_geometry::PinholeCameraModel::projectPixelTo3dRay](http://docs.ros.org/en/diamondback/api/image_geometry/html/c++/classimage__geometry_1_1PinholeCameraModel.html#a1f5a3957c06a249e250e74513319f8a8) to get a ray from each pixel (x.y). Beware that the function doesn't return a unit vector, but a vector with `z=1` to easy multiply standard depth values. In your case, you have to normalize the ray, then multiply by your depth. The resulting z component will be the actual depth that should be set in the depth image at the corresponding pixel.

closed 2024-02-01 by MotorTrick