introlab/rtabmap #1214: Kinect v1 CameraOpenNI2: Cannot start depth and/or color streams.

minystreem56, 2024-02-05:
When I pressed start it gave me this error using OpenNI2:
`CameraOpenNI2: Cannot start depth and/or color streams.`

minystreem56, 2024-02-08:
Any sugestions on how to fix?

matlabbe, 2024-02-11:
Is it on Windows? Latest release? What is the model of you kinect (written under it: e.g., `model 1414`)?

minystreem56, 2024-02-12:
Latest release, windows 11, not sure what model it is right now,
currently away from home, but I will check and report back later today

minystreem56, 2024-02-13:
My model is 1414

matlabbe, 2024-02-18:
I reproduced the same error on my Windows 11 desktop, but it works on my Windows 10 laptop. I am afraid that kinect xbox 360 compatibility died with windows 11: https://answers.microsoft.com/en-us/xbox/forum/all/my-windows-11-pc-will-not-recognize-my-xbox-360/39f345bd-dd55-496c-9594-f89806e82435

minystreem56, 2024-02-18:
thats sad

closed 2024-02-19 by matlabbe