introlab/rtabmap #1221: how to use superglue superpoints

qetuo105487900, 2024-02-19:
sorry to bother everyone, Superglue can run rtabslam without docker ? as follow step is correct?

roslaunch realsense2_camera rs_camera.launch align_depth:=true
roslaunch rtabmap_launch rtabmap.launch args:="-d
--SuperPoint/ModelPath /home/lun/rtabmap/archive/2022-IlluminationInvariant/scripts/superpoint_v1.pt
--SuperGlue/Path /home/lun/rtabmap/archive/2022-IlluminationInvariant/scripts/SuperGluePretrainedNetwork/demo_superglue.py
--Reg/RepeatOnce false
--Vis/CorGuessWinSize 0
--Kp/DetectorStrategy 11
--Vis/FeatureType 11
--Mem/UseOdomFeatures false
--Vis/CorNNType 6"
rtabmap_args:="--delete_db_on_start"
depth_topic:=/camera/aligned_depth_to_color/image_raw
rgb_topic:=/camera/color/image_raw
camera_info_topic:=/camera/color/camera_info
approx_sync:=false
i get this : QAQ
Features2d.cpp:594::create() SupertPoint Torch feature cannot be used as RTAB-Map is not built with the option enabled. GFTT/ORB is used instead.

matlabbe, 2024-02-19:
Yes, you can run without docker, but you will need to install cuda + pytorch (from source), the build rtabmap library with libtorch (for superpoint) and python (for superglue) support (`cmake -DWITH_TORCH=ON -DWITH_PYTHON=ON ..`).

qetuo105487900, 2024-02-21:
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5") found components: C CXX 
-- Found OpenMP: /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.10;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-7.1;/usr/include/freetype2;/usr/include/x86_64-linux-gnu
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL2
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter Development NumPy Development.Module Development.Embed 
-- Found Python3
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /opt/ros/noetic/include (found version "2.10.3")
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o (targets)
-- g2o include dir: /opt/ros/noetic/include
-- Old g2o factory version detected without shared ptr (factory file: /opt/ros/noetic/include/g2o/core/factory.h).
-- GTSAM include directory:  /opt/ros/noetic/lib/cmake/GTSAM/../../../include
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found Boost: /usr/include (found version "1.71.0") found components: thread filesystem system program_options date_time chrono atomic 
-- Found libpointmatcher: /opt/ros/noetic/include;/usr/include/eigen3
-- Found Boost: /usr/include (found version "1.71.0") found components: thread filesystem system program_options date_time chrono timer serialization atomic 
-- Found RealSense2: 
-- Found octomap 1.9.8: /opt/ros/noetic/include
-- Found grid_map_core 1.6.4: /opt/ros/noetic/include;/usr/include/eigen3
-- Architecture: x86_64
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.4
--   CMAKE_VERSION =        3.28.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++14
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_VERSION = 1.10.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.2.0 xfeatures2d = NO, nonfree = NO (License: BSD)
--   With Qt 5.12.8            = YES (License: Open Source or Commercial)
--   With VTK 7.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (libtorch not found)
--   With Python3.8            = YES (License: PSF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o 1.0.0            = YES (License: BSD)
--  *With GTSAM 4.2.0          = YES (License: BSD)
--  *With Ceres                = NO (WITH_CERES=OFF)
--   With MRPT                 = NO (MRPT not found)
--   With VERTIGO              = YES (License: GPLv3)
[... 21 lines cut ...]
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2 2.50.0    = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done (2.5s)
-- Generating done (0.3s)
-- Build files have been written to: /home/lun/rtabmap/build 

but With SupertPoint still is NO, who can help me!!! thx!!

qetuo105487900, 2024-02-21:
i change CMakeList.txt in rtabmap
such as this  :
 
IF(WITH_TORCH)
    set(Torch_DIR /home/lun/libtorch/share/cmake/Torch)
    FIND_PACKAGE(Torch REQUIRED)
    IF(TORCH_FOUND)
        MESSAGE(STATUS "Found Torch: ${TORCH_INCLUDE_DIRS}")
    ENDIF(TORCH_FOUND)
ENDIF(WITH_TORCH)

and it find !!!

qetuo105487900, 2024-02-21:
cmake -DWITH_TORCH=ON -DWITH_PYTHON=ON ..

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP: /usr/lib/gcc/x86_64-linux-gnu/9/libgomp.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/include/opencv4
-- Found PCL: /usr/include/pcl-1.10;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-7.1;/usr/include/freetype2;/usr/include/x86_64-linux-gnu
-- Found ZLIB: /usr/include
-- VTK_RENDERING_BACKEND=OpenGL2
-- Caffe2: CUDA detected: 12.2
-- Caffe2: CUDA nvcc is: /usr/local/cuda/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda
-- Caffe2: Header version is: 12.2
-- /usr/local/cuda/lib64/libnvrtc.so shorthash is 79a3de5a
-- USE_CUDNN is set to 0. Compiling without cuDNN support
-- USE_CUSPARSELT is set to 0. Compiling without cuSPARSELt support
-- Autodetected CUDA architecture(s):  8.6
-- Added CUDA NVCC flags for: -gencode;arch=compute_86,code=sm_86
-- Found Torch: /home/lun/libtorch/include;/home/lun/libtorch/include/torch/csrc/api/include
-- Found Python3
-- Found pybind11: /opt/ros/noetic/include (found version "2.10.3")
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o (targets)
-- g2o include dir: /opt/ros/noetic/include
-- Old g2o factory version detected without shared ptr (factory file: /opt/ros/noetic/include/g2o/core/factory.h).
-- GTSAM include directory:  /opt/ros/noetic/lib/cmake/GTSAM/../../../include
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- Found Boost: /usr/include (found version "1.71.0") found components: thread filesystem system program_options date_time chrono atomic 
-- Found libpointmatcher: /opt/ros/noetic/include;/usr/include/eigen3
-- Found Boost: /usr/include (found version "1.71.0") found components: thread filesystem system program_options date_time chrono timer serialization atomic 
-- Found RealSense2: 
-- Found octomap 1.9.8: /opt/ros/noetic/include
-- Found grid_map_core 1.6.4: /opt/ros/noetic/include;/usr/include/eigen3
-- Architecture: x86_64
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.4
--   CMAKE_VERSION =        3.28.3
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0  -fopenmp -std=c++17
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = -DDISABLE_OPENNI2;-DDISABLE_PCAP;-DDISABLE_PNG;-DDISABLE_LIBUSB_1_0
--   PCL_VERSION = 1.10.0
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.2.0 xfeatures2d = NO, nonfree = NO (License: BSD)
--   With Qt 5.12.8            = YES (License: Open Source or Commercial)
--   With VTK 7.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = YES (License: GPLv3) libtorch=2.2.0
--   With Python3.8            = YES (License: PSF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
-- 
--  Solvers:
[... 25 lines cut ...]
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2 2.50.0    = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done (2.7s)
-- Generating done (0.4s)
-- Build files have been written to: /home/lun/rtabmap/build

qetuo105487900, 2024-02-21:
rtabmap --version

without everything that i had do

RTAB-Map:               0.21.3
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With OctoMap:             true
With GridMap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false

matlabbe, 2024-02-24:
You can also do:
```
cmake -DTorch_DIR=/home/lun/libtorch/share/cmake/Torch ..
```
to avoid hard-coding it in the `CMakeLists.txt`.

The cmake status shows that torch is detected, so if rtabmap builds succesfully, it should have it. 

After build is complted, you must install `sudo make install`. It seems `rtabmap --version` is not your version you just built, but the one from the binaries. I suggest also to uninstall the binaries, as you will also need to [rebuild rtabmap_ros from source](https://github.com/introlab/rtabmap_ros/?tab=readme-ov-file#build-from-source):
```bash
# uninstall binaries
sudo apt remove ros-$ROS_DISTRO-rtabmap*

# install rtabmap with torch support
cd rtabmap/build
cmake -DTorch_DIR=/home/lun/libtorch/share/cmake/Torch ..
make -j6
sudo make install

# install rtabmap_ros with local rtabmap version:
cd ~/catkin_ws
git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
catkin_make -j4
```

anath93, 2024-03-13:
@matlabbe my platform is jetson (running jetpack 5) so I built libtorch using this link [wheel way](https://docs.nvidia.com/deeplearning/frameworks/install-pytorch-jetson-platform/index.html) and built rtabmap using
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DWITH_TORCH=ON -DWITH_PYTHON=ON -DTorch_DIR=/home/amr/.local/lib/python3.8/site-packages/torch/share/cmake/Torch ..

Using libtorch version of torch-2.1.0a0+41361538.nv23.06-cp38-cp38-linux_aarch64
But I am getting this error while querying rtabmap --version libc10.so and there was no error while build
[image]
Then Finally building rtabmap_ros breaks at

anath93, 2024-03-13:
@matlabbe I got superpoint to build with rtabmap, but does not like Python3 is getting recognized, I have hardcoded cmake file for now to find torch folder. 
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DWITH_TORCH=ON -DWITH_PYTHON=ON
[image]

closed 2024-05-09 by qetuo105487900