introlab/rtabmap #1222: rtabmap.db database file

takahiko-hasegawa-hdjp, 2024-02-23:
Hi there.

I have a question about the database file. After creating a map and use the map for localization mode, it seems like the .db file is opened and writing something to it and then closing it. so when if i see the details of the map.db(diff) there is some diff but not a lot.(so the timestamp becomes different). Is there a way so that the map is not modified everytime you load the map?

matlabbe, 2024-02-24:
By default, most actions would call rtabmap.close() with databaseSaved=true, even in localization mode
https://github.com/introlab/rtabmap/blob/81e249b56c516b016bc8fe2b14eda116c892a2ef/corelib/src/Rtabmap.cpp#L460-L462

In localization mode, if you are not using memory management,  here would be at least that line that would modify the database every time it is closed:
https://github.com/introlab/rtabmap/blob/81e249b56c516b016bc8fe2b14eda116c892a2ef/corelib/src/Rtabmap.cpp#L518

Under ROS, we also save the current optimized occupancy grid:
https://github.com/introlab/rtabmap_ros/blob/d7d2c3f122af955ea500d3cc49a4e718edc84a83/rtabmap_slam/src/CoreWrapper.cpp#L879-L893

If you are not on ROS, you may just add this inside `Rtabmap::close()` function:
```
databaseSaved = databaseSaved && _memory->isIncremental();
```
I think then the database won't be modified, unless I forget another place.

takahiko-hasegawa-hdjp, 2024-02-25:
Ok Just to clarify a few things;

I can modify the code like you said in order for the database to not be modified after running rtabmap. Im using ros2 rtabmap. 

If i modify the code like you said, would that make any differece to the actual localization mode like it wont localize as good? or the performance?

Also, the odometry correction of the robot would be the position of the robot from the previous session. 
So run the rtabmap node -> mode the robot -> stop the robot->map-to-odom would be the same as the previous map to odom. 
If I disable the database save function as mentioned above, would I still be able use the "re-loading" of the previous map-to-odom feature?

I do want to change the code so that the database will not be modified, but I do not want to make any difference to the actual system performance at all. 

Thanks!

matlabbe, 2024-02-26:
If we don't save the last location position in the database, on restart, the robot won't have a any a priori knowledge where it is restarting to help first localization. It will behave like of if the robot is shutdown, moved to another place and booted up again. The first localization guess will be wrong, thus it would rely only on global localization. In most cases, it won't change anything, as global localization could work on the first frame or two if the robot is seeing enough discriminating visual features.

The map->odom correction is always adjusted on start depending on the new odometry origin and where the robot thinks it is in the map on start.

Not saving the back the last position in the database is similar to use `RGBD/StartAtOrigin`:
`Param: RGBD/StartAtOrigin = "false"                        [If true, rtabmap will assume the robot is starting from origin of the map. If false, rtabmap will assume the robot is restarting from the last saved localization pose from previous session (the place where it shut down previously). Used only in localization mode (Mem/IncrementalMemory=false).]`

This won't change localization performance, just where the robot thinks it is on start.

takahiko-hasegawa-hdjp, 2024-02-26:
That makes perfect sense. thanks for your answer!

closed 2024-02-27 by matlabbe