introlab/rtabmap #1228: IMU + RGBD image drift

maxmoeschinger, 2024-03-01:
Hi,

I am currently setting up rtabmap in ROS with a RGB camera + 3d lidar sensor + IMU.

My problem is that the map is drifting after just a few seconds. See image below (supposed to be a wall on one line but it is curving down)
[image]

My launch file looks like this:
```
<?xml version="1.0"?>
<launch>
    <arg name="frame_id"                default="base_link" />
    <arg name="rpi_camera"              default="false" />
    <arg name="base_folder"             default="/app" />
    <arg name="left_lidar_id"           default="A03E-2530-2CC3-3A70" />
    <arg name="right_lidar_id"          default="9C3E-253B-2CC3-3A6C" />
    <arg name="lidar_slave"             default="false" />
    <arg name="rosbag_path"             default="" />
    <arg name="record_rosbag"           default="false" />
    <arg name="robot_localization"      default="false" />

    <!-- Setup Lidar geometry and transforms -->
    <param name="robot_description" textfile="$(arg base_folder)/launch/config/lidar-2.0.0.xml"/>
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="rob_st_pub" >
        <remap from="robot_description" to="robot_description" />
        <remap from="joint_states" to="different_joint_states" />
    </node>

    <!-- Nodelet managers -->
    <node pkg="nodelet" type="nodelet" name="left_lidar_manager" args="manager" output="screen" />

    <group ns="input">
        <group ns="rosbag" unless="$(eval '' == rosbag_path)">
            <node pkg="rosbag" type="play" name="player" output="screen" args="--clock $(arg rosbag_path)"/>
        </group>
        <group ns="record_rosbag" if="$(arg record_rosbag)">
            <node pkg="rosbag" type="record" name="rosbag_record" args="record -o /app/data/rosbag /input/sensor/imu/data /input/sensor/imu/magnetometer /input/sensor/rpi_camera_v3_wide_384_216/camera_info /input/sensor/rpi_camera_v3_wide_384_216/image_raw /royale_cam_left_lidar/camera_info /royale_cam_left_lidar/depth_image_0 /royale_cam_left_lidar/point_cloud_0"/>
        </group>
        <group ns="sensor" if="$(eval '' == rosbag_path)">
            <!-- Video input -->
            <include file="$(find video_stream_opencv)/launch/camera.launch" >
                <!-- node name and ros graph name -->
                <arg name="camera_name" value="rpi_camera_v3_wide_384_216" />
                <!-- means video device 0, /dev/video0 -->
                <arg name="video_stream_provider" value="http://127.0.0.1:5000" />
                <!-- set buffer queue size of frame capturing to -->
                <arg name="buffer_queue_size" value="1" />
                <!-- throttling the publishing of frames to -->
                <arg name="fps" value="40" />
                <!-- setting frame_id -->
                <arg name="frame_id" value="elp_camera" />
                <!-- camera info loading, take care as it needs the "file:///" at the start , e.g.:
                "file:///$(find your_camera_package)/config/your_camera.yaml" -->
                <arg name="camera_info_url" value="file:///app/launch/config/rpi_camera_v3_wide_384_216.yaml" />
                <!-- flip the image horizontally (mirror it) -->
                <arg name="flip_horizontal" value="false" />
                <!-- flip the image vertically -->
                <arg name="flip_vertical" value="false" />
            </include>
            <!-- Lidar input -->
            <node pkg="nodelet" type="nodelet" name="left_lidar_driver" args="load royale_in_ros/royale_nodelet /left_lidar_manager" respawn="true" output="screen">
                <param name="node_name"          type="str"    value="left_lidar"/>
                <param name="royale_cam_id"  type="str"      value="$(arg left_lidar_id)" />
                <param name="royale_use_case"  type="str"      value="Long_Range_30FPS" />
            </node>
            <!-- IMU -->
            <node name="ros_imu_bno055_node" pkg="ros_imu_bno055" type="imu_ros.py" output="screen">
                <param name="serial_port" type="string" value="/dev/ttyAMA0" />
                <param name="frame_id" type="string"    value="base_link" />
                <param name="use_magnetometer"          value="true" />
[... 100 lines cut ...]
            imu_msg.orientation.x,
            imu_msg.orientation.y,
            imu_msg.orientation.z,
            imu_msg.orientation.w,
        ))
        imu_msg.orientation.x = quat_norm[0]
        imu_msg.orientation.y = quat_norm[1]
        imu_msg.orientation.z = quat_norm[2]
        imu_msg.orientation.w = quat_norm[3]

        imu_msg.orientation_covariance = covariance_matrix(0.001)
        imu_msg.linear_acceleration_covariance = covariance_matrix(1)
        imu_msg.angular_velocity_covariance = covariance_matrix(0.1)
        imu_orientation_quaternion_pub.publish(imu_msg)

    rospy.Subscriber("imu/data_raw", Imu, process_new_imu_msg)

    try:
        while not rospy.is_shutdown():
            time.sleep(1)
    except rospy.ROSInterruptException:
        pass
```

Do you have any idea why I can't achieve good results with rtabmap?

matlabbe, 2024-03-03:
Can you share a rosbag of the following topics?
```
rosbag record /tf /tf_static \
   /input/sensor/rpi_camera_v3_wide_384_216/image_rect_color \
   /input/sensor/rpi_camera_v3_wide_384_216/camera_info \
   /input/sensor/rpi_camera_v3_wide_384_216/depth_image_0_gen_rect \
   /input/sensor/imu/data_processed \
   /royale_cam_left_lidar/point_cloud_0
```

Here are some stuff that can go wrong:
 * Low or different frame rate between the RGB and depth cameras
 * Bad synchronization of the rgb and depth frames
 * Bad TF between the RGB and depth cameras (how extrinsics have been computed?)
 * ...

maxmoeschinger, 2024-03-04:
Yes, here you go: https://drive.google.com/file/d/1bxXn9blpWZCKAzm6dPpivV9nQd-p6DIQ/view?usp=sharing

I have only done manual calibration of the RGB and depth camera. Do you have a good library to do the extrinsic calibration?

matlabbe, 2024-03-05:
Can you have an IR stream from the depth camera? If so, you may be do something like this: http://wiki.ros.org/openni_launch/Tutorials/ExtrinsicCalibrationExternal

It seems there is also a module in opencv to do that: https://docs.opencv.org/4.x/d2/d1c/tutorial_multi_camera_main.html

For the data in the rosbag, thetime difference between RGB and depth frames is sometime too large:
```
The time difference between rgb and depth frames is high (diff=0.133255s, rgb=1709536481.863311s, depth=1709536481.730056s).
```
It is a problem when the camera is moving, the depth image won't be correctly registered to color image. On Google Tango we had the same issue (depth image frame rate is 5 Hz and rgb camera is 30 Hz), but with Tango VIO computed at 30 Hz, we could correctly register the depth cloud to corresponding rgb frame.

The depth is not dense enough for feature extraction, either reduce the depth resolution by two (to make it more dense) or fill the holes:
[image]

For such small resolution, you can set `GFTT/MinDistance` to 3.

For reference, I tested the bag with:
```
roslaunch rtabmap_launch rtabmap.launch \
   args:="-d --GFTT/MinDistance 3 --Rtabmap/DetectionRate 0" \
   rgb_topic:=/input/sensor/rpi_camera_v3_wide_384_216/image_rect_color \
   camera_info_topic:=/input/sensor/rpi_camera_v3_wide_384_216/camera_info \
   depth_topic:=/input/sensor/rpi_camera_v3_wide_384_216/depth_image_0_gen_rect  \
   use_sim_time:=true \
   frame_id:=elp_camera \
   queue_size:=50
```

maxmoeschinger, 2024-03-05:
Thank you so much for such good support. I am already getting way better results by increasing the density of the density image.
One thing I am wondering about is why my `odom` frame is rotating compared to the `map` frame. It starts off really good where both odom and map frame are almost the same:
[image]

But it then suddenly rotates and shifts:
[image]
Shouldn't the IMU gravity constraint how the odom frame is allowed to move with regards to the map frame? No mather how bad the relation between odom and map is it seems like the relation between base_link and odom is correct(orientation and position).

matlabbe, 2024-03-09:
Do you delete the database on each restart (argument `--delete_db_on_start`)? It looks like it is relocalizing on a previous session. Otherwise, map -> odom would change on loop closure detection. In rtabmap_viz you would see a green background on loop closure view when it happens. Another thing that could happen is if the base_link->IMU transform is wrong, the map optimizer may wrong optimize the map with IMU constraints. You can turn off IMU optimization to compare if it is the case by setting `--Optimizer/GravitySigma 0`.