introlab/rtabmap #1229: use image filter replace original color image but not work!

qetuo105487900, 2024-03-03:
i use this command 

```
roslaunch rtabmap_launch rtabmap.launch \
    rtabmap_args:="--delete_db_on_start" \
    depth_topic:=/rs_l515/aligned_depth_to_color/image_raw \
    rgb_topic:=/rs_l515/color/image_filter \
    camera_info_topic:=/rs_l515/color/camera_info \
    approx_sync:=false
```
is not work on rtabmap

but i use this command 
```
roslaunch rtabmap_launch rtabmap.launch \
    rtabmap_args:="--delete_db_on_start" \
    depth_topic:=/rs_l515/aligned_depth_to_color/image_raw \
    rgb_topic:=/rs_l515/color/image_raw \
    camera_info_topic:=/rs_l515/color/camera_info \
    approx_sync:=false
```

it can work on rtabmap

so,  how to do ? i can run with image_filter topic? copy camera info ?

qetuo105487900, 2024-03-03:
always get the warning code
```
[ WARN] [1709477545.228805757]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/odom \
   /filter/rs_l515/color/image_raw \
   /filter/rs_l515/aligned_depth_to_color/image_raw \
   /filter/rs_l515/color/camera_info \
   /rtabmap/odom_info

```

qetuo105487900, 2024-03-03:
i see here:
https://robotics.stackexchange.com/questions/23394/rtabmap-rtabmap-did-not-receive-data-since-5-seconds

mobiiin say is encoding problem ? is it right ?

qetuo105487900, 2024-03-03:
i also run `rostopic hz`
and i found the original and the filters' are very similar. so maybe is not the problem.

qetuo105487900, 2024-03-04:
should i use tf replace tf_static ?

matlabbe, 2024-03-05:
What is the rostopic hz output for:
```
rostopic hz \
   /filter/rs_l515/color/image_raw \
   /filter/rs_l515/aligned_depth_to_color/image_raw \
   /filter/rs_l515/color/camera_info \
```

qetuo105487900, 2024-03-05:
hello, @matlabbe 

[1] 
run with 
    depth_topic:=/rs_l515/aligned_depth_to_color/image_raw \
    rgb_topic:=/rs_l515/color/image_raw \
    camera_info_topic:=/rs_l515/color/camera_info \
which publish from realsense nodelet
[image]
this can work well ! 

[2] 
run with 
    depth_topic:=/filter/rs_l515/aligned_depth_to_color/image_raw \
    rgb_topic:=/filter/rs_l515/color/image_raw \
    camera_info_topic:=/filter/rs_l515/color/camera_info \
which publish from (realsense nodelet after sharping then publish)
[image]

this can't work  ! 

[3]
this is hz between 
/filter/rs_l515/color/image_raw 
/rs_l515/color/image_raw /filter/rs_l515/aligned_depth_to_color/image_raw
/rs_l515/aligned_depth_to_color/image_raw
[image]

qetuo105487900, 2024-03-05:
maybe problem is at the tf not publish from rs_l515_link?

matlabbe, 2024-03-09:
You should have warnings in rtabmap terminal if topics can be synchronized but don't have the right format or if there is a TF issue. If you have warnings about topics not received for 5 seconds, you may check if `/filter/rs_l515/color/camera_info` also published.

I see that you are also using `approx_sync:=false`, if the topics don't have the same stamps, use `approx_sync:=true` instead.

qetuo105487900, 2024-03-10:
yes, mr. @matlabbe , i had use `approx_sync:=true` instead. 
but i still can't slove the problem. 
so, i can't use `/rs_l515/color/camera_info` as camera_info_topic ?
like this :
```
  <!-- RGB-D related topics -->
  <arg name="rgb_topic"               default="filter/rs_d435i_up/color/image_raw" />
  <arg name="depth_topic"             default="/rs_d435i_up/aligned_depth_to_color/image_raw" />
  <arg name="camera_info_topic"       default="/rs_d435i_up/color/camera_info" />
  <arg name="depth_camera_info_topic" default="$(arg camera_info_topic)" />
```

as below is sharpened code, are there any problem in the code ?

```
def color_callback(data):
    global laplacian_kernel
    try:
        cv_image = bridge.imgmsg_to_cv2(data, "bgr8")
    except CvBridgeError as e:
        print(e)

    # 局部亮度调整
    sharpened_img = sharpened(cv_image)

    # 將BGR圖像轉換為RGB圖像
    sharpened_img = cv2.cvtColor(sharpened_img, cv2.COLOR_BGR2RGB)

    try:
        deflicker_sharpened_msg = bridge.cv2_to_imgmsg(sharpened_img, "rgb8")

        # Publishing
        pub_filter_color.publish(deflicker_sharpened_msg)

        
    except CvBridgeError as e:
        print(e)

if __name__ == '__main__':
    rospy.init_node('image_filter', anonymous=True)

    bridge = CvBridge()
    # Subscribers
    rospy.Subscriber("/rs_d435i_up/color/image_raw", Image, color_callback)
    
    # Publishers
    pub_filter_color = rospy.Publisher("/filter/rs_d435i_up/color/image_raw", Image, queue_size=10)
    rospy.spin()

```

matlabbe, 2024-03-13:
You can use `/rs_d435i/color/camera_info` if `/filter/rs_d435i/color/image_raw` has still the same resolution (and that you didn't change the scale of the image, i.e., kept same calibration).

matlabbe, 2024-03-13:
Another thing I just notice, you should copy the old header in your new published topic header.
```
sharpened_img.header = data.header
```

closed 2024-05-09 by qetuo105487900