introlab/rtabmap #1232: expected type-specifier before ‘OdometryORBSLAM’  : ORBSLAM2  RTAB-MAP.

zhuoyan28, 2024-03-05:
I installed ORBSLAM2 in conjunction with RTAB-MAP.
I get this error when RTAB-MAP executes make.
It looks like it's a problem with gcc g++.
In ORBSLAM2 patch (from https://zhuanlan.zhihu.com/p/426604435)
The reason for the error when executing the patch directly is that the changes to the CXX in the three CMakeLists are more or less the same, with only a few differences, and ORBSLAM2 compiles without problems.

install reference
https://zhuanlan.zhihu.com/p/528557998
https://zhuanlan.zhihu.com/p/426604435

the output: 
[ 33%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o
[ 33%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerCVSBA.cpp.o
[ 33%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerCeres.cpp.o
[ 34%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Registration.cpp.o
[ 34%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationIcp.cpp.o
[ 35%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationVis.cpp.o
[ 35%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Odometry.cpp.o
/home/master/study-learning/rtabmap/corelib/src/Odometry.cpp: In static member function ‘static rtabmap::Odometry* rtabmap::Odometry::create(rtabmap::Odometry::Type&, const ParametersMap&)’:
/home/master/study-learning/rtabmap/corelib/src/Odometry.cpp:89:18: error: expected type-specifier before ‘OdometryORBSLAM’
   odometry = new OdometryORBSLAM(parameters);
                  ^~~~~~~~~~~~~~~
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1650: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/Odometry.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/Odometry.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

zhuoyan28, 2024-03-08:
Avoid the problem.       Do not download the latest rtab-map version, select the suffix melodic (ubuntu18), and ros corresponding to this version, you will not encounter this problem when runningm, choose anth

matlabbe, 2024-03-09:
The ORB_SLAM integration has been updated to support ORB_SLAM2 and ORB_SLAM3 in that commit https://github.com/introlab/rtabmap/commit/aaff1abc4f257cf948de5f427d190a89556510ac. Based on your error, there is indeed a bug here:
https://github.com/introlab/rtabmap/blob/52e1b015978d7f02b139d4e598a8bd49fe98fd17/corelib/src/Odometry.cpp#L88-L92
It should be:
```cpp
#if defined(RTABMAP_ORB_SLAM) and RTABMAP_ORB_SLAM == 2
		odometry = new OdometryORBSLAM2(parameters);
#else
		odometry = new OdometryORBSLAM3(parameters);
#endif
```

matlabbe, 2024-03-09:
Fixed in commit above. I tested on ubuntu 20.04 / OpenCV 4, so I used this PR: https://github.com/raulmur/ORB_SLAM2/pull/1075 with these local changes:
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 465e9ff..7bdab9c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,8 @@ ENDIF()
 
 MESSAGE("Build type: " ${CMAKE_BUILD_TYPE})
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -O3 ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -O3")
 
 # Check c++14 or C++0x support
 include(CheckCXXCompilerFlag)
diff --git a/Thirdparty/DBoW2/CMakeLists.txt b/Thirdparty/DBoW2/CMakeLists.txt
index e20f340..a174009 100644
--- a/Thirdparty/DBoW2/CMakeLists.txt
+++ b/Thirdparty/DBoW2/CMakeLists.txt
@@ -1,8 +1,8 @@
 cmake_minimum_required(VERSION 2.8)
 project(DBoW2)
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3  ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 ")
 
 set(HDRS_DBOW2
   DBoW2/BowVector.h
diff --git a/Thirdparty/g2o/CMakeLists.txt b/Thirdparty/g2o/CMakeLists.txt
index 620e688..c1a3c91 100644
--- a/Thirdparty/g2o/CMakeLists.txt
+++ b/Thirdparty/g2o/CMakeLists.txt
@@ -54,8 +54,8 @@ IF(OPENMP_FOUND AND G2O_USE_OPENMP)
 ENDIF(OPENMP_FOUND AND G2O_USE_OPENMP)
 
 # Compiler specific options for gcc
-SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native") 
-SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native") 
+SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 ") 
+SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 ")
```

closed 2024-03-09 by matlabbe

zhuoyan28, 2024-03-10:
Thank you for your reply. My environment is ubuntu18. Now that ORBSLAM2 has been installed, which version of rtab-map should be installed later? Directly git the latest version or the --melodic version, such as 0.20.23-melodic?

zhuoyan28, 2024-03-10:
My environment dependencies are：
ubuntu 18.04
ros-melodic-desktop-full
pangolin 0.5
opcv 3.2 (provided by ros-melodic)

zhuoyan28, 2024-03-10:
1 Thank you for your reply. My environment is ubuntu18. Now that ORBSLAM2 has been installed, which version of rtab-map should be installed later? Directly git the latest version or the --melodic version, such as 0.20.23-melodic?

2  My environment dependencies are：
ubuntu 18.04
ros-melodic-desktop-full
pangolin 0.5
opcv 3.2 (provided by ros-melodic)

3  When compiling ORBSLAM2, I directly git the version of ORBSLAM2, and then use orbslam2_f2e6f51_marchnative_disabled.patch: and then compile again.

The orbslam2_f2e6f51_marchnative_disabled.patch:  
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 08a8af4..8190289 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,8 @@ ENDIF()
 
 MESSAGE("Build type: " ${CMAKE_BUILD_TYPE})
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   ")
 
 # Check C++11 or C++0x support
 include(CheckCXXCompilerFlag)
diff --git a/Thirdparty/DBoW2/CMakeLists.txt b/Thirdparty/DBoW2/CMakeLists.txt
index 0eb5126..db34af2 100644
--- a/Thirdparty/DBoW2/CMakeLists.txt
+++ b/Thirdparty/DBoW2/CMakeLists.txt
@@ -1,8 +1,8 @@
 cmake_minimum_required(VERSION 2.8)
 project(DBoW2)
 
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall ")
 
 set(HDRS_DBOW2
   DBoW2/BowVector.h
diff --git a/Thirdparty/g2o/CMakeLists.txt b/Thirdparty/g2o/CMakeLists.txt
index 620e688..ba8396c 100644
--- a/Thirdparty/g2o/CMakeLists.txt
+++ b/Thirdparty/g2o/CMakeLists.txt
@@ -54,8 +54,8 @@ IF(OPENMP_FOUND AND G2O_USE_OPENMP)
 ENDIF(OPENMP_FOUND AND G2O_USE_OPENMP)
 
 # Compiler specific options for gcc
-SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native") 
-SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native") 
+SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} ") 
+SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE}") 
 
 # activate warnings !!!
 SET(g2o_C_FLAGS "${g2o_C_FLAGS} -Wall -W")

zhuoyan28, 2024-03-10:
1 Thank you for your reply. My environment is ubuntu18. Now that ORBSLAM2 has been installed, which version of rtab-map should be installed later? Directly git the latest version or the --melodic version, such as 0.20.23-melodic?

2 My environment dependencies are：
ubuntu 18.04
ros-melodic-desktop-full
pangolin 0.5
opcv 3.2 (provided by ros-melodic)

3 When compiling ORBSLAM2, I directly git the version of ORBSLAM2, and then use orbslam2_f2e6f51_marchnative_disabled.patch: and then compile again.

The orbslam2_f2e6f51_marchnative_disabled.patch:
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 08a8af4..8190289 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -7,8 +7,8 @@ ENDIF()

MESSAGE("Build type: " ${CMAKE_BUILD_TYPE})

-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall ")

Check C++11 or C++0x support
include(CheckCXXCompilerFlag)
diff --git a/Thirdparty/DBoW2/CMakeLists.txt b/Thirdparty/DBoW2/CMakeLists.txt
index 0eb5126..db34af2 100644
--- a/Thirdparty/DBoW2/CMakeLists.txt
+++ b/Thirdparty/DBoW2/CMakeLists.txt
@@ -1,8 +1,8 @@
cmake_minimum_required(VERSION 2.8)
project(DBoW2)

-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -O3 -march=native ")
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -O3 -march=native")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall ")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall ")

set(HDRS_DBOW2
DBoW2/BowVector.h
diff --git a/Thirdparty/g2o/CMakeLists.txt b/Thirdparty/g2o/CMakeLists.txt
index 620e688..ba8396c 100644
--- a/Thirdparty/g2o/CMakeLists.txt
+++ b/Thirdparty/g2o/CMakeLists.txt
@@ -54,8 +54,8 @@ IF(OPENMP_FOUND AND G2O_USE_OPENMP)
ENDIF(OPENMP_FOUND AND G2O_USE_OPENMP)

Compiler specific options for gcc
-SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native")
-SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native")
+SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} ")
+SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE}")

activate warnings !!!
SET(g2o_C_FLAGS "${g2o_C_FLAGS} -Wall -W")

zhuoyan28, 2024-03-10:
I completed all the compilation and ran roslaunch and this problem occurred. （rtab-map  and rtab-map-ros both are obtained directly from github）

roslaunch rtabmap_ros euroc_datasets.launch args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/shen/rtab_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt" MH_seq:=true raw_images_for_odom:=true
RLException: [euroc_datasets.launch] is neither a launch file in package [rtabmap_ros] nor is [rtabmap_ros] a launch file name
The traceback for the exception was written to the log file

matlabbe, 2024-03-13:
Hi,

For your previous message, you should be able to build latest master branch on 18.04. 

The euroc_datasets.launch is now under rtabmap_examples package:
```
roslaunch rtabmap_examples euroc_datasets.launch \
   args:="Odom/Strategy 5 OdomORBSLAM/VocPath /home/shen/rtab_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt" \
   MH_seq:=true \
   raw_images_for_odom:=true
```

Tested with version [above](https://github.com/introlab/rtabmap/issues/1232#issuecomment-1987006668) and I could not reproduce the error. Is it happening at the beginning or later in the bag? (to know at least it is somewhat working)

My log just after starting the bag:
```
Camera Parameters: 
- fx: 436.235
- fy: 436.235
- cx: 364.445
- cy: 256.952
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 3
- Scale Factor: 2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 4.40312
New map created with 360 points
[ INFO] [1710303025.539362217, 1403636579.864527989]: Odom: quality=360, std dev=99.995000m|99.995000rad, update time=0.026084s
[ WARN] (2024-03-12 21:10:25.541) MainWindow.cpp:1162::processOdometry() Initializing rectification maps for stereo camera 0 (only done for the first image received)...
[ WARN] (2024-03-12 21:10:25.543) MainWindow.cpp:1165::processOdometry() Initializing rectification maps for stereo camera 0 (only done for the first image received)... done!
[ WARN] (2024-03-12 21:10:25.545) Memory.cpp:4617::createSignature() Initializing rectification maps (only done for the first image received)...
[ WARN] (2024-03-12 21:10:25.547) Memory.cpp:4619::createSignature() Initializing rectification maps (only done for the first image received)...done!
[ INFO] [1710303025.565914547, 1403636579.894708984]: Odom: quality=315, std dev=0.013760m|0.010000rad, update time=0.025260s
[ INFO] [1710303025.576130198, 1403636579.904766035]: rtabmap (1): Rate=1.00s, Limit=0.000s, Conversion=0.0035s, RTAB-Map=0.0241s, Maps update=0.0006s pub=0.0054s (local map=1, WM=1)
[ INFO] [1710303025.601048830, 1403636579.924941348]: Odom: quality=341, std dev=0.013760m|0.010000rad, update time=0.017992s
[ INFO] [1710303025.645353533, 1403636579.975226139]: Odom: quality=315, std dev=0.013760m|0.010000rad, update time=0.016947s
[ INFO] [1710303025.699690301, 1403636580.025600847]: Odom: quality=328, std dev=0.013760m|0.010000rad, update time=0.016873s
[ INFO] [1710303025.759981157, 1403636580.085991395]: Odom: quality=325, std dev=0.013760m|0.010000rad, update time=0.017051s
[ INFO] [1710303025.798672590, 1403636580.126225381]: Odom: quality=309, std dev=0.013760m|0.010000rad, update time=0.016956s
[ INFO] [1710303025.844623470, 1403636580.166457297]: Odom: quality=300, std dev=0.013760m|0.010000rad, update time=0.016153s
[ INFO] [1710303025.894758915, 1403636580.216736663]: Odom: quality=248, std dev=0.013760m|0.010000rad, update time=0.016127s
[ INFO] [1710303025.944100258, 1403636580.267044496]: Odom: quality=229, std dev=0.013760m|0.010000rad, update time=0.015587s
[ INFO] [1710303025.995424869, 1403636580.317345581]: Odom: quality=267, std dev=0.013760m|0.010000rad, update time=0.016958s
[ INFO] [1710303026.043645073, 1403636580.367643672]: Odom: quality=227, std dev=0.013760m|0.010000rad, update time=0.015215s
[ INFO] [1710303026.093691921, 1403636580.417978663]: Odom: quality=266, std dev=0.013760m|0.010000rad, update time=0.015304s
[ INFO] [1710303026.143575674, 1403636580.468365728]: Odom: quality=264, std dev=0.013760m|0.010000rad, update time=0.015215s
[ INFO] [1710303026.196837418, 1403636580.518748492]: Odom: quality=175, std dev=0.013760m|0.010000rad, update time=0.014708s
[ INFO] [1710303026.242466293, 1403636580.569031047]: Odom: quality=193, std dev=0.013760m|0.010000rad, update time=0.014019s
[ INFO] [1710303026.297040695, 1403636580.619319190]: Odom: quality=142, std dev=0.013760m|0.010000rad, update time=0.013711s
[ INFO] [1710303026.346424544, 1403636580.669606476]: Odom: quality=135, std dev=0.013760m|0.010000rad, update time=0.013305s
[ INFO] [1710303026.396769740, 1403636580.719926996]: Odom: quality=239, std dev=0.013760m|0.010000rad, update time=0.016937s
[ INFO] [1710303026.446012837, 1403636580.770233603]: Odom: quality=260, std dev=0.013760m|0.010000rad, update time=0.017306s
[ INFO] [1710303026.497988774, 1403636580.820535579]: Odom: quality=297, std dev=0.013760m|0.010000rad, update time=0.019433s
[ INFO] [1710303026.527875896, 1403636580.850719649]: rtabmap (2): Rate=1.00s, Limit=0.000s, Conversion=0.0006s, RTAB-Map=0.0279s, Maps update=0.0001s pub=0.0010s (local map=2, WM=2)
[ INFO] [1710303026.550168325, 1403636580.870827638]: Odom: quality=322, std dev=0.013760m|0.010000rad, update time=0.021450s
[ INFO] [1710303026.601209024, 1403636580.925210528]: Odom: quality=365, std dev=0.013760m|0.010000rad, update time=0.018969s
```

zhuoyan28, 2024-03-15:
Sorry to bother you, my previous installation was to directly git the latest rtab-map and rtab-map-ros codes.
If I just use the new Ubuntu system and follow the installation docker file, I can install both rtab-map + ORBSLAM2, right?

matlabbe, 2024-03-16:
Yeah, you can also create a docker to run on any ubuntu os.

zhuoyan28, 2024-03-17:
thank you so much！

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/17/2024 07:22 |
| To | introlab/rtabmap ***@***.***> |
| Cc | Dan ***@***.***>,
Author ***@***.***> |
| Subject | Re: [introlab/rtabmap] expected type-specifier before ‘OdometryORBSLAM’ : ORBSLAM2 RTAB-MAP. (Issue #1232) |

Yeah, you can also create a docker to run on any ubuntu os.

commit 2577c47e (2024-03-09): Fixed #1232