introlab/rtabmap #1235: G2go Fatal error for optimizeBA()

jenadzitsiuk, 2024-03-09:
`[FATAL] (2024-03-08 20:07:10.475) OptimizerG2O.cpp:1835::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!`

I'm getting this error at the start of Rtabmap reconstruction Ubuntu standalone app:
* G2o is most recent master
* PCL built from source - 1.13
* GSTM is not installed. If I do add it, the rtabmap GUI does not even start - it crashes right away
* Running with reallsense2 cameras - D435i.
* tried to build with most recent ceres solver - rtabmap does not build

Are there any ways I can troubleshoot?
Also what versions of G2o, GSTM, ceres, PCL, VTK, Realsense2 correspond to most recent Rtabmap (0.21.4)? May be I have some conflicts?

Sometimes the rtabmap does start and starts showing warnings about IMU orientation not computed or some warnings from OdometryF2M.cpp about registation failed, but eventually it goes to some ftatal optimizeBA error. Not sure if those are related.
```

My cmake output:
-- Info :
--   RTAB-Map Version =     0.21.4
--   CMAKE_VERSION =        3.22.1
--   CMAKE_INSTALL_PREFIX = /usr/local
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0 -fopenmp -std=c++17
--   FLANN_KDTREE_MEM_OPT = OFF
--   PCL_DEFINITIONS = 
--   PCL_VERSION = 1.13.1.99
--   PCL_COMPILE_OPTIONS = -mavx2
-- 
-- Optional dependencies ('*' affects some default parameters) :
--  *With OpenCV 4.7.0 xfeatures2d = NO, nonfree = NO (License: Apache 2)
--   With Qt 5.15.3            = YES (License: Open Source or Commercial)
--   With VTK 9.1              = YES (License: BSD)
--   With external SQLite3     = YES (License: Public Domain)
--   With ORB OcTree           = YES (License: GPLv3)
--   With SupertPoint          = NO (WITH_TORCH=OFF)
--   With Python3              = NO (WITH_PYTHON=OFF)
--   With Madgwick             = YES (License: GPL)
--   With FastCV               = NO (FastCV not found)
--   With PDAL                 = NO (PDAL not found)
-- 
--  Solvers:
--   With TORO                 = YES (License: Creative Commons [Attribution-NonCommercial-ShareAlike])
--  *With g2o 1.0.0            = YES (License: BSD)
--  *With GTSAM                = NO (WITH_GTSAM=OFF)
--  *With Ceres                = NO (WITH_CERES=OFF)
--   With MRPT                 = NO (MRPT not found)
--   With VERTIGO              = YES (License: GPLv3)
--   With cvsba                = NO (WITH_CVSBA=OFF)
--  *With libpointmatcher      = NO (libpointmatcher not found)
--   With CCCoreLib            = NO (WITH_CCCORELIB=OFF)
--   With Open3D               = NO (WITH_OPEN3D=OFF)
--   With OpenGV               = NO (OpenGV not found)
-- 
--  Reconstruction Approaches:
--   With OctoMap              = NO (octomap not found)
--   With GridMap              = NO (grid_map_core not found)
--   With CPUTSDF              = NO (WITH_CPUTSDF=OFF)
--   With OpenChisel           = NO (WITH_OPENCHISEL=OFF)
--   With AliceVision          = NO (WITH_ALICE_VISION=OFF)
-- 
--  Camera Drivers:
--   With Freenect             = NO (libfreenect not found)
--   With OpenNI2              = YES (License: Apache v2)
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2 2.54.2    = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jenna/dev/rtabmap/build
```

matlabbe, 2024-03-09:
Which gtsam version are you using? the version now available though ros should work (`ros-$ROS_DISTRO-gtsam`).

ceres v2 build error is a known issue: https://github.com/introlab/rtabmap/issues/1219

I cannot reproduce the error with latest g2o during a quick test with a D435i. How long does it take for the error to happen? Can you enable the Debug log in Preferences to show log till it asserts. I see you have many dependencies built from source. In general, rtabmap is well tested with default binaries version of these dependencies that you can install on the ubuntu version / ros version you are using.

jenadzitsiuk, 2024-03-11:
Thanks for a quick answer!
I did re-install PCL.  uninstalled ceres. 
Now PCL is not built from source, but 1.12 installed as "sudo apt install libpcl-dev"
GTSAM not installed (this is not necessary right?). No other changes compared to my original post.

Good news, the rtabmap app is working now. I see the 3D map growing and no errors in the console
But if I start the rgbd-camera app, its crashing. Could that be an issue with VTK?
I wonder why I see differences in 3D rendering execution of rtabmap and rtabmap-rgbd_camera, they use different code for rendering? 

call stack:
```
#0  0x00007f6230f19c10 in _XEventsQueued () from /lib/x86_64-linux-gnu/libX11.so.6
[Current thread is 1 (Thread 0x7f62137fe640 (LWP 140498))]
(gdb) bt
#0  0x00007f6230f19c10 in _XEventsQueued () at /lib/x86_64-linux-gnu/libX11.so.6
#1  0x00007f6230f06291 in XPending () at /lib/x86_64-linux-gnu/libX11.so.6
#2  0x00007f622c06db8f in vtkXRenderWindowInteractor::StartEventLoop() () at /lib/x86_64-linux-gnu/libvtkRenderingUI-9.1.so.1
#3  0x00007f623fd7df8c in pcl::visualization::PCLVisualizer::spinOnce(int, bool) () at /lib/x86_64-linux-gnu/libpcl_visualization.so.1.12
#4  0x00007f623fda42f0 in pcl::visualization::CloudViewer::CloudViewer_impl::operator()() () at /lib/x86_64-linux-gnu/libpcl_visualization.so.1.12
#5  0x00007f623dedc253 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f623da94ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#7  0x00007f623db26850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```
log:
```
[ INFO] (2024-03-11 11:55:20.478) main.cpp:186::main() Using driver 11 (device=0)
[ INFO] (2024-03-11 11:55:20.605) CameraRealSense2.cpp:490::init() setupDevice...
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:517::init() Device "Intel RealSense D435I" with serial number 153122074883 was found with product ID=2874.
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:608::init() Using device with Serial No: 153122074883
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:611::init() Device Name: Intel RealSense D435I
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:614::init() Device FW version: 5.12.7.150
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:617::init() Device Product ID: 0x0B3A
[ INFO] (2024-03-11 11:55:20.613) CameraRealSense2.cpp:627::init() Device Sensors: 
[ INFO] (2024-03-11 11:55:20.614) CameraRealSense2.cpp:692::init() Stereo Module was found.
[ INFO] (2024-03-11 11:55:20.614) CameraRealSense2.cpp:692::init() RGB Camera was found.
[ INFO] (2024-03-11 11:55:20.614) CameraRealSense2.cpp:692::init() Motion Module was found.
[ INFO] (2024-03-11 11:55:20.614) CameraRealSense2.cpp:705::init() Sensor 0 "RGB Camera"
[ INFO] (2024-03-11 11:55:20.616) CameraRealSense2.cpp:708::init() profiles=224
[ INFO] (2024-03-11 11:55:20.616) CameraRealSense2.cpp:714::init() RGB8 1920 1080 30 0 Color type=2
[ INFO] (2024-03-11 11:55:20.616) CameraRealSense2.cpp:714::init() Y16 1920 1080 30 0 Color type=2
[ INFO] (2024-03-11 11:55:20.616) CameraRealSense2.cpp:714::init() Y8 1920 1080 30 0 Color type=2
.....
[ INFO] (2024-03-11 11:55:20.618) CameraRealSense2.cpp:714::init() BGR8 320 180 6 0 Color type=2
[ INFO] (2024-03-11 11:55:20.618) CameraRealSense2.cpp:714::init() YUYV 320 180 6 0 Color type=2
[ INFO] (2024-03-11 11:55:20.618) CameraRealSense2.cpp:756::init() Model: 640x480 fx=605.897583 fy=605.802612 cx=324.214630 cy=252.450745 dist model=2 coeff=0.000000 0.000000 0.000000 0.000000 0.000000
[ INFO] (2024-03-11 11:55:20.618) CameraRealSense2.cpp:705::init() Sensor 1 "Stereo Module"
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:708::init() profiles=108
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:714::init() Y8 1280 800 30 1 Infrared 1 type=3
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:714::init() Y16 1280 800 25 1 Infrared 1 type=3
.......
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:714::init() Z16 424 240 6 0 Depth type=1
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:714::init() Z16 256 144 300 0 Depth type=1
[ INFO] (2024-03-11 11:55:20.620) CameraRealSense2.cpp:714::init() Z16 256 144 90 0 Depth type=1
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:705::init() Sensor 2 "Motion Module"
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:708::init() profiles=4
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:714::init() MOTION_XYZ32F 0 0 250 0 Accel type=6
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:714::init() MOTION_XYZ32F 0 0 63 0 Accel type=6
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:714::init() MOTION_XYZ32F 0 0 400 0 Gyro type=5
[ INFO] (2024-03-11 11:55:20.631) CameraRealSense2.cpp:714::init() MOTION_XYZ32F 0 0 200 0 Gyro type=5
[ INFO] (2024-03-11 11:55:20.632) CameraRealSense2.cpp:959::init() imu local transform = xyz=-0.011305,-0.020265,0.005370 rpy=-1.572469,0.021230,-1.562441
[ INFO] (2024-03-11 11:55:20.632) CameraRealSense2.cpp:1072::init() Starting sensor 0 with 1 profiles
[ INFO] (2024-03-11 11:55:20.632) CameraRealSense2.cpp:1076::init() Opening: RGB8 640 480 30 0 Color type=2
[ INFO] (2024-03-11 11:55:20.632) CameraRealSense2.cpp:1088::init() Set RS2_OPTION_GLOBAL_TIME_ENABLED=1 (was 1.000000) for sensor 0
[ INFO] (2024-03-11 11:55:20.634) CameraRealSense2.cpp:1072::init() Starting sensor 1 with 1 profiles
[ INFO] (2024-03-11 11:55:20.634) CameraRealSense2.cpp:1076::init() Opening: Z16 640 480 30 0 Depth type=1
[... cut ...]
2024-03-11 11:55:21.826 (   1.355s) [        98FF9640]vtkOpenGLPolyDataMapper:321   WARN| vtkOpenGLPolyDataMapper::GetFragmentShaderCode was deprecated for VTK 9.0 and will be removed in a future version.  Use vtkOpenGLShaderProperty::GetFragmentShaderCode instead.
2024-03-11 11:55:21.826 (   1.355s) [        98FF9640]vtkOpenGLPolyDataMapper:313   WARN| vtkOpenGLPolyDataMapper::SetFragmentShaderCode was deprecated for VTK 9.0 and will be removed in a future version.  Use vtkOpenGLShaderProperty::SetFragmentShaderCode instead.
Segmentation fault (core dumped)
```

matlabbe, 2024-03-14:
`rtabmap-rgbd_camera` is using the PCL visualizer directly (without Qt), while `rtabmap` is using our own viewer that is also based on PCLVisualizer but works under Qt. 

There could be something buggy between PCL1.12 and VTK9.1. My current system is running PCl 1.10 and VTK 7.1 and no problems. Just tried on 22.04 and I can reproduce your error with `rtabmap-rgbd_camera`, here is the backtrace from gdb:
```
Thread 51 "rtabmap-rgbd_ca" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff677fe640 (LWP 636)]
0x00007fffdfd06c10 in _XEventsQueued () from /lib/x86_64-linux-gnu/libX11.so.6
(gdb) bt
#0  0x00007fffdfd06c10 in _XEventsQueued () at /lib/x86_64-linux-gnu/libX11.so.6
#1  0x00007fffdfcf3291 in XPending () at /lib/x86_64-linux-gnu/libX11.so.6
#2  0x00007fffdf610b8f in vtkXRenderWindowInteractor::StartEventLoop() () at /lib/x86_64-linux-gnu/libvtkRenderingUI-9.1.so.1
#3  0x00007ffff425ff8c in pcl::visualization::PCLVisualizer::spinOnce(int, bool) ()
    at /lib/x86_64-linux-gnu/libpcl_visualization.so.1.12
#4  0x00007ffff42862f0 in pcl::visualization::CloudViewer::CloudViewer_impl::operator()() ()
    at /lib/x86_64-linux-gnu/libpcl_visualization.so.1.12
#5  0x00007ffff3725253 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff3492ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#7  0x00007ffff3524850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```
Not sure if it is the same error that you get though, I ran this under docker, but the main app `rtabmap` is correctly working. It seems the only workaround to use PCL CloudViewer is to rebuild PCL from source: https://github.com/PointCloudLibrary/pcl/issues/5237

Beware that when your rebuild pcl from source is that if PCL uses `-march=native` flag, all other libraries using Eigen like g2o or gtsam should also use it to avoid runtime crashes.

jenadzitsiuk, 2024-03-18:
Great thank you so much! Rebuilding PCL is a reasonable approach, thanks for linking the bug on PCL issue tracker!

closed 2024-03-18 by jenadzitsiuk