introlab/rtabmap #1236: rtabmap parameter problem in 0.21.4

qetuo105487900, 2024-03-10:
i see rtabmap_ros/rtabmap_demos/demo_husky.launch

there are some parameter can not found in RTAB-Map:               0.21.4
but i found some similar parameter, if there are same parameter ?

Icp/PMOutlierRatio vs Icp/OutlierRatio
Icp/PointToPlaneNormalK vs Icp/PointToPlaneK

matlabbe, 2024-03-13:
You can use the old names and they will work, but you will have a warning message to know to which new ones you should use.

The new name for `Icp/PMOutlierRatio` is `Icp/OutlierRatio`:
https://github.com/introlab/rtabmap/blob/57b4954cb355efe2dc8f5035896d82f7977868bc/corelib/src/Parameters.cpp#L257

For `Icp/PointToPlaneNormalK`, I think there is an error, it should indeed be `Icp/PointToPlaneK`. The default is 5 while the husky demo set it to 10, maybe why that bug passed unnoticed. I [fixed](https://github.com/introlab/rtabmap_ros/commit/665735663ef7bf029c11c599c06a1fd727299e0c) the typo.

qetuo105487900, 2024-03-13:
@matlabbe , thanks!

closed 2024-03-13 by qetuo105487900