introlab/rtabmap #1237: how to change 2d map when i navigate from 1f to 2f

qetuo105487900, 2024-03-11:
excuse me. i have a problem about changing 2d map from 1f to 2f.

i have a amr. it will use joystick to make amr start at 1f a point then take a elevater to 2f b point.

Do I just need to shut down the node and then restart it ?
or rtabmap have some service can clear current map and load new map ?

matlabbe, 2024-03-13:
You may create 2 databases, one for each floor. You can either kill the node and reload with other database, or call service `rtabmap/load_database`.

If you want to keep both floors in same database, after mapping the first floor you call service `rtabmap/pause` in the elevator, rotate the robot so it doesn't see the same last image of the first floor session, then call `rtabmap/trigger_new_map` and `rtabmap/resume` to continue mapping. Map the second floor. After the run, you will have a single database with 2 disjoints maps in side. On localization mode, the robot would switch between the maps implicitly after exiting the elevator. The occupancy grid map should be updated based on the map the robot is now in.

qetuo105487900, 2024-03-13:
@matlabbe , thank you so much !!!!!

matlabbe, 2024-03-16:
In your example, I assume that you already created two databases, one for each floor. If you know on which floor the robot should be, you can indeed call `rtabmap/load_database` with the corresponding database.

matlabbe, 2024-04-28:
If you call `rtabmap/load_database`, it would change the path of the database used by rtabmap node. For the pose, the robot may jump to latest location saved in that database. You may override the pose by sending a new pose on `/rtabmap/initialpose` topic.

qetuo105487900, 2024-05-07:
ok . i see . thank you so much !

closed 2024-05-07 by qetuo105487900