introlab/rtabmap #1238: Empty Gridmap and Octomap with recent integration of Gridmap

FlorianMehnert, 2024-03-11:
Hi there,

with the integration of `grid_map_integration` into the master #1180 neither Octomap or Gridmap generation work anymore for me anymore. Gridmap publishes Maps containing only Nan values and Octomap publishes empty maps.

Do you have any idea on why this could happen?

As a side effect RviZ is crashing with: `catkin_ws/src/grid_map/grid_map_core/src/GridMap.cpp:45: void grid_map::GridMap::setGeometry(const Length&, double, const Position&): Assertion `length(0) > 0.0' failed.`

matlabbe, 2024-03-13:
Just tried the latest rtabmap noetic binary releases 0.21.4 with this [example](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Stereo_mapping):
```
roslaunch rtabmap_demos demo_stereo_outdoor.launch rviz:=true
rosbag play --clock stereo_outdoorA.bag
```

and I can get the three topics at the same time:
[image]

I see grid_map is built from source, are you also using rtabmap from source?

FlorianMehnert, 2024-03-13:
Yes I was using rtabmap from source. After 2 days of doing nothing everything works :sweat_smile: 
Maybe there was an issue with the OAK camera I am using or something else :)

closed 2024-03-13 by FlorianMehnert