introlab/rtabmap #1241: Python3 Option with RTABMAP Issue

anath93, 2024-03-14:
I have been trying to build Rtabmap with Python3 and torch support but Python3 seems not be coming true, cleared workspace multiple times and re-build with 
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release -DWITH_TORCH=ON -DWITH_PYTHON=ON

The result is :
[image]

[image]

anath93, 2024-03-15:
I tried using command, 
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DWITH_TORCH=ON -DWITH_PYTHON=ON -DCMAKE_CXX_FLAGS="-I/usr/include/python3.8/"
Still cannot build rtabmap with Python3 support.

Torch builds only after passing path to shared library  like below otherwise Rtabmap still builds then after after checking rtabmap --version, it complains about libc10.so which is torch shared library object not in path.
export LD_LIBRARY_PATH=/home/amr/.local/lib/python3.8/site-packages/torch/lib:"$LD_LIBRARY_PATH"

Until here standard KNN + superpoint could be used but not superglue + superpoint.

Any help would be appreciated, platform is Ubuntu 20.04 + ROS2 Galactic + 2.1.0 Torch + Jetpack 5.1.2

matlabbe, 2024-03-17:
This is how cmake searches for python:
https://github.com/introlab/rtabmap/blob/264c560a2c43d7abbc0563a05d96adf77841ba9c/CMakeLists.txt#L401

You can add a REQUIRED so that cmake shows more info when python is not found:
```cmake
FIND_PACKAGE(Python3 COMPONENTS Interpreter Development NumPy REQUIRED) 
```

If cmake cannot find it, you may search on google to get more help.

Vis/RoiRatios value sould be a string: '0.0 0.0 0.2 0.3'

anath93, 2024-03-17:
@matlabbe thank you for the reply, I will close this case. For now appending this line made Python3 compile with RTABMAP but still not the right solution but I will look into it, 
FIND_PACKAGE(Python3 3.8.10 EXACT REQUIRED COMPONENTS Interpreter Development)
[image]

closed 2024-03-17 by anath93

anath93, 2024-03-18:
if anyone has this issue, the problem is with symlink of python libraries, Nvidia uses 3.9 for there side and ROS uses 3.8, so removing the symlink for 3.9 will work to build 
ls -l /usr/bin | grep python
then remove sudo rm /usr/bin/pythonX.X (desired one)