introlab/rtabmap #1242: Support for Orbbec Femto cameras

tonhathuy, 2024-03-15:
Thank you, this project is great!, 
Is there any plans to support orbbec Femto cameras?

matlabbe, 2024-03-16:
If I get my hands on one for sure!

jeddors, 2024-06-17:
just fyi, Orbbec has a K4a wrapper (https://github.com/orbbec/OrbbecSDK-K4A-Wrapper) which lets you use the Femto cameras as if they were the Azure Kinect

Replace/add the dll's into RTAB:
k4a.dll
k4arecord.dll
OrbbecSDK.dll
depthengine_2_0.dll

I've only tested this with the Femto Bolt

CatoFuyu, 2024-07-22:
Hello Jedd,

I came across this thread while looking for ways to integrate the Femto Bolt with RTAB-Map.
Thank you for the valuable information.
Could you please provide more details on how to set up the camera?
I have copied the four .dll files to the execution folder, but I still get the message "Camera initialization failed...".
The Femto Bolt is recognized in the Device Manager.

Thank you

Kato

jeddors, 2024-07-22:
> Hello Jedd,
>
> I came across this thread while looking for ways to integrate the Femto Bolt with RTAB-Map. Thank you for the valuable information. Could you please provide more details on how to set [...]

First make sure that your Femto Bolt is working correctly with the default Orbbec drivers to rule out any external issues

I'm guessing you followed the install instructions and made changes only to the  bin folder of the K4A sdk (...\Program Files\Azure Kinect SDK v1.4.1)

From what I can see, RTAB seems to come bundled with its own K4A libraries so you need to copy the files into the .../RTAB/bin directory too:
depthengine_2_0.dll
k4a.dll
k4arecord.dll
OrbbecSDK.dll

This got it working for me. 
I didn't really bother to find out if copying all four wrapper .dll is necessary.
Overwriting the K4A sdk is probably also not required since its included with RTAB.

CatoFuyu, 2024-07-23:
Hello Jedd,

Thank you for your prompt response.

To confirm the steps, you mentioned copying the following files from the OrbbecSDK_K4A_Wrapper_v1.10.1_windows_2024_05_14/bin folder:

depthengine_2_0.dll
k4a.dll
k4arecord.dll
OrbbecSDK.dll
to both the Program Files/Azure Kinect SDK v1.4.2/tools folder and the rtabmap/build/bin folder.

I tried this, but I'm still getting the "Camera initialization failed..." message.

Additionally, do I need to input the camera ID and other information in the Source tab of the RTAB-Map Preferences, or does RTAB-Map automatically recognize the camera?

My computer has no issues recognizing the Femto Bolt. The tests in Orbbec Viewer run without any problems, and it appears correctly in the Device Manager.

Please lend me a hand.

Thank you.
[image]

jeddors, 2024-07-23:
> Hello Jedd,
> 
> Thank you for your prompt response. I’m extremely grateful for your help
> 
> To confirm the steps, you mentioned copying the following files from the OrbbecSDK_K4A_Wrapper_v1.10.1_windows_2024_05_14/bin folder:
> 
> depthengine_2_0.dll k4a.dll k4arecord.dll OrbbecSDK.dll to both the Program Files/Azure Kinect SDK v1.4.2/tools folder and the rtabmap/build/bin folder.
> 
> I tried this, but I'm still getting the "Camera initialization failed..." message.
> 
> Additionally, do I need to input the camera ID and other information in the Source tab of the RTAB-Map Preferences, or does RTAB-Map automatically recognize the camera?
> 
> My computer has no issues recognizing the Femto Bolt. The tests in Orbbec Viewer run without any problems, and it appears correctly in the Device Manager.
> 
> Please lend me a hand.
> 
> Thank you.
> 
> Cato
> 
> ![source_pic](https://private-user-images.githubusercontent.com/35454215/351175693-9afa5aba-9318-48e4-a746-aa6c78ab8f1d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE3NTcwOTcsIm5iZiI6MTcyMTc1Njc5NywicGF0aCI6Ii8zNTQ1NDIxNS8zNTExNzU2OTMtOWFmYTVhYmEtOTMxOC00OGU0LWE3NDYtYWE2Yzc4YWI4ZjFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIzVDE3NDYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRlZTMwZjAwYTZjYTRmYTg1YjQ4MjE0NzllYTVhMmNmOTVlYTA5ZGEwMjA2M2IxOWVjNzU3YjhmNGY0MzljYjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sct1sE8tbwZM57oMYZ4iDliePUYB678KzWSUNWse-Vk)

To install the K4A wrapper on the SDK, you should follow guide.

You can confirm whether you've done this correctly by accessing the Femto Bolt through the Azure Kinect Viewer successfully. 

After placing the wrapper DLLs into the RTAB/bin directory, you don't need to do any further config to initialize the camera just:
Detection -> Select Source -> RGB-D camera -> Kinect for Azure

CatoFuyu, 2024-07-24:
Sure! Here is the translation:

Hello Jedd,

Finally, RTAB-Map on my computer has recognized the Femto Bolt!
I want to thank you once again for your helpful comments.
If I hadn't found this thread, I might have given up.

I wish you all the best.

Thank you

Kato

emvnuelrm, 2025-02-25:
Hello!
I have little knowledge of how these open source SDKs work, the only thing I want is to be able to scan places. Is Rtabmap used to obtain a file from a scan using the Femto Mega?
I would greatly appreciate your response.
Greetings!