introlab/rtabmap #1243: what are these lines mean in graph view ?

qetuo105487900, 2024-03-15:
[image]
i know blue line is my slam path 
but how about the yellow and the red lines mean?

matlabbe, 2024-03-16:
Yellow is proximity detection (or local loop closure detection), red is global loop closure detection. If you right-click on the view, there is a menu option showing the different link colors (that can be changed). To know more about those links, see section 3.4 of this paper: [RTAB-Map as an Open-Source Lidar and Visual SLAM Library for Large-Scale and Long-Term Online Operation](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf).

qetuo105487900, 2024-03-17:
@matlabbe , thanks!!!!

closed 2024-03-17 by qetuo105487900