introlab/rtabmap #1244: "Is it possible to map multiple times using the same parameters? Then, combine the results of multiple maps together, ensuring that localization is preserved in subsequent processing."

qetuo105487900, 2024-03-15:
"Is it possible to map multiple times using the same parameters? Then, combine the results of multiple maps together, ensuring that localization is preserved in subsequent processing."

qetuo105487900, 2024-03-16:
i use rgbd-camera 3d lidar and odometry

matlabbe, 2024-03-17:
You can do:
```
rtabmap-reprocess "map1.db;map2.db" combined_maps.db 
```

qetuo105487900, 2024-03-17:
@matlabbe , thanks!!!!!!!!!!

qetuo105487900, 2024-04-22:
@matlabbe , sorry. Should I need to start at the same start point ?

matlabbe, 2024-04-28:
No, you don't need to, but the robot should traverse at some point same locations than in previous session, ideally with same point of view so that a loop closure can be detected. It is then easier to just start from the same position.

qetuo105487900, 2024-05-07:
ok, i see. thank you so much

closed 2024-05-07 by qetuo105487900