introlab/rtabmap #1250: `cmake_modules/FindSqlite3.cmake` breaks transitive dependencies.

dg0yt, 2024-03-23:
Since CMake 3.14, there is an official `FindSQLite3.cmake`, variable prefix `SQLite3`, and an imported target.
rtabmap comes with `FindSqlite3.cmake`, variable prefix `Sqlite3`.
On case-insensitive filesystems (Windows), this can break finding downstream dependencies with `find_dependency(SQLite3)`. This was noticed in vcpkg when updating PROJ, via a rtabmap > opencv > vtk > proj > sqlite3 dependency chain.

~~~
CMake Error at C:/a/1/s/scripts/buildsystems/vcpkg.cmake:859 (_find_package):
  Found package configuration file:

    D:/installed/x64-windows/share/proj/proj-config.cmake

  but it set PROJ_FOUND to FALSE so package "PROJ" is considered to be NOT
  FOUND.  Reason given by package:

  PROJ could not be found because dependency SQLite3 could not be found.

Call Stack (most recent call first):
  D:/installed/x64-windows/share/vtk/VTK-vtk-module-find-packages.cm	820 (find_package)
  D:/installed/x64-windows/share/vtk/vtk-config.cmake:150 (include)
  C:/a/1/s/scripts/buildsystems/vcpkg.cmake:859 (_find_package)
  D:/downloads/tools/cmake-3.27.1-windows/cmake-3.27.1-windows-i386/share/cmake-3.27/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  D:/installed/x64-windows/share/opencv/OpenCVModules.cmake:43 (find_dependency)
  D:/installed/x64-windows/share/opencv/OpenCVConfig.cmake:174 (include)
  C:/a/1/s/scripts/buildsystems/vcpkg.cmake:859 (_find_package)
  CMakeLists.txt:232 (FIND_PACKAGE)
~~~

Possible solutions:
- Move to a more distinct module name.
- Switch to the official module. Needs CMake 3.14.
  Tentative patch in https://github.com/microsoft/vcpkg/pull/37650/files#diff-0a2a39d0c34a319ba9d49bc64e1da201038a1311b2e18368cab394a84353d439

matlabbe, 2024-03-23:
I implemented the second suggestion. You could try this tag [0.21.4.1](https://github.com/introlab/rtabmap/releases/tag/0.21.4.1) on your [pull request](https://github.com/microsoft/vcpkg/pull/37650).

dg0yt, 2024-03-28:
I didn't update my PR which discovered the problem, but I'm building successfully with 0.21.4.1 in a more general review of the vcpkg port, https://github.com/microsoft/vcpkg/pull/37651.