introlab/rtabmap #1257: Continously updating the reconstructed cloud with the sensor field of view

aasishkc4, 2024-04-05:
Hi,
I have a robot equipped with a 3D lidar and a depth camera and I have used both of the sensors for 3d reconstruction. Now I wanted the reconstructed cloud to keep updating the scene which is currently in the field of view of the sensors. How can I achieve that? Currently, the cloud gets updated when the head moves which is as the camera and lidar are attached to the head.

matlabbe, 2024-04-06:
If only the head is moving, but not the base frame, you could set `RGBD/LinearUpdate` to 0 to force updating the map all the time. Currently, rtabmap may think the robot is immobile, so it doesn't update the map.

aasishkc4, 2024-04-08:
Sure. I will try that. One more doubt, many times I get this tf error any thoughts on that?

[WARN] [1712557006.951025452] [rtabmap]: We received odometry message, but we cannot get the corresponding TF world->camera_depth_optical_frame at data stamp 251.932000s (odom msg stamp is 251.956000s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.

closed 2024-04-08 by aasishkc4

matlabbe, 2024-04-10:
You would need to look at the `/rtabmap/octomap...` outputs. To make OctoMap clear dynamic obstacles, you need to set `Grid/RayTracing` to true (with `Grid/3D=true` if not already).

For the tf warnings, are you in simulation? The delay is 20 ms, if it doesn't appear often, you may ignore it.