introlab/rtabmap #1259: Offload the processing of the RTAB-Map iOS app to a PC

nirmalsnair, 2024-04-09:
Hello,

Is there a way to offload the processing of the RTAB-Map iOS app to a PC?

I'm looking to establish a server capable of receiving input data from one or more remote devices (such as smartphones, robots, etc.). Currently, as a temporary solution, we've made slight modifications to the iOS code. During scanning, we spawn a background thread every 8 seconds to export the 3D model and transmit it to our server. At the server end, we utilize a babylon.js web browser to display the 3D model. However, this approach still relies on processing within the iPhone and tends to crash when scanning large areas.

What would you suggest to facilitate such a client-server system?
- Modify the standalone app to enable remote device connections instead of USB connections
- Modify the iOS app to better suit our requirements
- Use RTAB-Map ROS (though I'm not sure how this could be integrated with iPhones)
- Explore alternative options?

PS: We are particularly fond of the iOS interface and would prefer to maintain a similar user experience if possible. Thanks!

matlabbe, 2024-04-11:
For your first point, remote mapping [can be done](http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping) through `rtabmap_ros` with on ros1 or ros2. We did [something similar](http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer) to what you described back in the days with Google Tango. For iOS, I don't think there is a project similar to Tango ROS Streamer, you would need to create your own ros node on iOS to stream ARKit data.

If you want to keep the same local visualization of the map on the phone like RTAB-Map iOS app, you may keep the app like that and add a function using ros1/ros2 library to stream the data outside on the network at the same time (I would only stream pose+image+camer_info at 1 hz to not flood the network). The remote rtabmap_ros node would do a mirror of what is happening on the phone, there would be no feedback from remote computer to phone.

For the multi-device mapping in real-time, it seems similar to https://github.com/introlab/rtabmap/issues/936#issuecomment-1333081446 and https://github.com/introlab/rtabmap/issues/1134

nirmalsnair, 2024-04-13:
Thanks for the comments.

> If you want to keep the same local visualization of the map on the phone like RTAB-Map iOS app, you may keep the app like that and add a function using ros1/ros2 library to stream the data outside [...]

In this case, does it imply that the same processing would be carried out on both the iPhone and the server as well?

matlabbe, 2024-04-13:
Yes. If you only want do the work on remote computer then send back the result to iPhone, you would need to add a callback from external to send a RtabmapEvent that would be [caught](https://github.com/introlab/rtabmap/blob/a55b4203cd9021a5b0b4e5cee188dfe410c59aab/app/android/jni/RTABMapApp.cpp#L4005-L4019) by the jni code without having an Rtabmap instance running on iPhone (or implemented in swift and data sent to c++ code, though I think it would be easier if ros1/ros2 interface is done in the c++ code instead of swift to avoid translating data structures between both).

nirmalsnair, 2024-04-14:
> though I think it would be easier if ros1/ros2 interface is done in the c++ code instead of swift to avoid translating data structures between both

In this case, will I be able to use the (modified) iOS app for the front end? We would prefer to retain RTAB-Map's iOS UI/UX over the ROS/Desktop app's UI/UX.

matlabbe, 2024-04-14:
You could do something like this (note that I didn't say it will be easy to do, especially if you don't know how the jni/c++ code is currently doing):
1. iOS app publishes : [rtabmap_msgs/RGBDImage](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/RGBDImage.html) + [nav_msgs/Odometry](http://docs.ros.org/en/noetic/api/nav_msgs/html/msg/Odometry.html), everything related to `rtabmap_` object would need to be removed if you don't want any mapping on the phone.
2. [rtabmap_slam/rtabmap](http://wiki.ros.org/rtabmap_slam#rtabmap) (no UI!) receives [rtabmap_msgs/RGBDImage](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/RGBDImage.html) + [nav_msgs/Odometry](http://docs.ros.org/en/noetic/api/nav_msgs/html/msg/Odometry.html) and update the map on its side (desktop computer)
3. [rtabmap_slam/rtabmap](http://wiki.ros.org/rtabmap_slam#rtabmap) publishes [rtabmap_msgs/MapData](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/MapData.html)
4. iOS app receives [rtabmap_msgs/MapData](http://docs.ros.org/en/noetic/api/rtabmap_msgs/html/msg/MapData.html) from ros, create an RtabmapEvent and publish it internally to current rendering loop of iOS app.

The current iOS UI would be used only for visualization, unless you remap every non-UI actions to ros services of rtabmap node. All the "Library" UI would need to be skipped as there would be no local database saved. Initializing a new scan would be done manually by starting the rtabmap ros node from a fresh database before starting the phone app, unless the appropriate ros services (to create a new database or load an existing database on the server side with rtabmap node always running) are connected between ios app and rtabmap node.

nirmalsnair, 2024-04-17:
Thank you for the detailed explanation.

closed 2024-04-17 by matlabbe