introlab/rtabmap #1265: Unable to launch GUI on Raspberry Pi 5

Nathan-Pattison, 2024-04-14:
> raspberrypi:~ $ rtabmap
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> [ WARN] (2024-04-14 13:41:21.197) Parameters.cpp:1251::readINI() Section "Core" in /home/nathan/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
> Program started...
> QWaylandGLContext: failed to create EGLContext, error=3009
> QOpenGLWidget: Failed to create context
> QWaylandGLContext: failed to create EGLContext, error=3009
> QOpenGLWidget: Failed to create context
> QWaylandGLContext: failed to create EGLContext, error=3009
> qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
> QWaylandGLContext: failed to create EGLContext, error=3009
> qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
> 

Is this an error in the RTAB-Map library or the Raspberry Pi 5's OpenGL graphics library?
I compiled RTAB-Map from source

matlabbe, 2024-04-14:
Which OS/version are you using?

Maybe related post: https://github.com/colmap/colmap/issues/749

Nathan-Pattison, 2024-04-15:
My Os/version is `Linux raspberrypi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64`

The linked post doesn't seem to relate to the issue

> nathan@raspberrypi:~ $ qmake -v
QMake version 3.1
Using Qt version 5.15.8 in /usr/lib/aarch64-linux-gnu

> nathan@raspberrypi:~ $ echo $QT_SELECTION
/usr/bin/qmake

> nathan@raspberrypi:~ $ rtabmap
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> [ WARN] (2024-04-15 16:49:49.770) Parameters.cpp:1251::readINI() Section "Core" in /home/nathan/.rtabmap/rtabmap.ini doesn't exist... Ignore this warning if the ini file does not exist yet. The ini file will be automatically created when rtabmap will close.
> Program started...
> QWaylandGLContext: failed to create EGLContext, error=3009
> QOpenGLWidget: Failed to create context
> QWaylandGLContext: failed to create EGLContext, error=3009
> QOpenGLWidget: Failed to create context
> QWaylandGLContext: failed to create EGLContext, error=3009
> qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
> QWaylandGLContext: failed to create EGLContext, error=3009
> qt.qpa.backingstore: composeAndFlush: makeCurrent() failed

matlabbe, 2024-04-17:
If you are okay with docker, you may try this: https://github.com/introlab/rtabmap_ros/issues/1141#issuecomment-2054324477 (same instructions should work on any arm32 or arm64)

However I don't recommend to visualize 3d stuff directly on the pi, but I would rather start rviz or rtabnapviz on a remote computer.

Nathan-Pattison, 2024-04-18:
Is there somewhere I can read about rviz/rtabnapviz remote visualisation, or perhaps codebases that use it? There seems to be decent documentation on the ROS side but I'm using the C++ library directly.

closed 2024-04-19 by Nathan-Pattison