introlab/rtabmap #1266: We received odometry message, but we cannot get the corresponding TF world->camera_depth_optical_frame at data stamp 0.449000s (odom msg stamp is 0.481000s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.

aasishkc4, 2024-04-23:
I am constantly getting this error 
`[rtabmap-1] [INFO] [1713848546.447671930] [rtabmap]: rtabmap(maps): latch                      = true
[rtabmap-1] [INFO] [1713848546.447886165] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-1] [INFO] [1713848546.447913757] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-1] [INFO] [1713848546.447923916] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-1] [INFO] [1713848546.447932707] [rtabmap]: rtabmap(maps): map_always_update          = true
[rtabmap-1] [INFO] [1713848546.447940757] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-1] [INFO] [1713848546.447948068] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-1] [INFO] [1713848546.447955627] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-1] [INFO] [1713848546.447962319] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-1] [INFO] [1713848546.448040962] [rtabmap]: rtabmap(maps): octomap_tree_depth         = 4
[rtabmap-1] [INFO] [1713848546.467859526] [rtabmap]: rtabmap: frame_id      = camera_depth_optical_frame
[rtabmap-1] [INFO] [1713848546.467899848] [rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-1] [INFO] [1713848546.467912685] [rtabmap]: rtabmap: log_to_rosout_level  = 4
[rtabmap-1] [INFO] [1713848546.467920971] [rtabmap]: rtabmap: initial_pose  = 
[rtabmap-1] [INFO] [1713848546.467927973] [rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-1] [INFO] [1713848546.467934588] [rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-1] [INFO] [1713848546.467946680] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-1] [INFO] [1713848546.467954084] [rtabmap]: rtabmap: odom_sensor_sync   = true
[rtabmap-1] [INFO] [1713848546.467961411] [rtabmap]: rtabmap: pub_loc_pose_only_when_localizing = false
[rtabmap-1] [INFO] [1713848546.467968245] [rtabmap]: rtabmap: gen_scan  = false
[rtabmap-1] [INFO] [1713848546.467975154] [rtabmap]: rtabmap: gen_depth  = false
[rtabmap-1] [INFO] [1713848546.471485257] [rtabmap]: rtabmap: scan_cloud_max_points = 0
[rtabmap-1] [INFO] [1713848546.471515293] [rtabmap]: rtabmap: scan_cloud_is_2d = false
[rtabmap-1] [INFO] [1713848546.491609123] [rtabmap]: Setting RTAB-Map parameter "Grid/Sensor"="2"
[rtabmap-1] [INFO] [1713848546.499399129] [rtabmap]: Setting RTAB-Map parameter "Optimizer/GravitySigma"="0"
[rtabmap-1] [INFO] [1713848546.501594975] [rtabmap]: Setting RTAB-Map parameter "Reg/Force3DoF"="true"
[rtabmap-1] [INFO] [1713848546.502015545] [rtabmap]: Setting RTAB-Map parameter "Rtabmap/DetectionRate"="10.0"
[rtabmap-1] [INFO] [1713848546.507106113] [rtabmap]: Setting "Icp/PointToPlaneRadius" parameter to 0 (default 0.000000) as "subscribe_scan_cloud" is true.
[rtabmap-1] [INFO] [1713848546.507271353] [rtabmap]: RTAB-Map detection rate = 10.000000 Hz
[rtabmap-1] [INFO] [1713848546.641481736] [rtabmap]: rtabmap: Deleted database "/home/svaya-ml/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[rtabmap-1] [INFO] [1713848546.641561447] [rtabmap]: rtabmap: Using database from "/home/svaya-ml/.ros/rtabmap.db" (0 MB).
[rtabmap-1] [INFO] [1713848546.692967679] [rtabmap]: rtabmap: Database version = "0.21.4".
[rtabmap-1] [INFO] [1713848546.693019118] [rtabmap]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[rtabmap-1] [INFO] [1713848546.702274779] [rtabmap]: Setup callbacks
[rtabmap-1] [INFO] [1713848546.702379674] [rtabmap]: rtabmap: subscribe_depth = true
[rtabmap-1] [INFO] [1713848546.702396865] [rtabmap]: rtabmap: subscribe_rgb = true
[rtabmap-1] [INFO] [1713848546.702405550] [rtabmap]: rtabmap: subscribe_stereo = false
[rtabmap-1] [INFO] [1713848546.702412515] [rtabmap]: rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmap-1] [INFO] [1713848546.702418730] [rtabmap]: rtabmap: subscribe_sensor_data = false
[rtabmap-1] [INFO] [1713848546.702423850] [rtabmap]: rtabmap: subscribe_odom_info = false
[rtabmap-1] [INFO] [1713848546.702428461] [rtabmap]: rtabmap: subscribe_user_data = false
[rtabmap-1] [INFO] [1713848546.702432973] [rtabmap]: rtabmap: subscribe_scan = false
[rtabmap-1] [INFO] [1713848546.702437552] [rtabmap]: rtabmap: subscribe_scan_cloud = true
[rtabmap-1] [INFO] [1713848546.702442132] [rtabmap]: rtabmap: subscribe_scan_descriptor = false
[rtabmap-1] [INFO] [1713848546.702447219] [rtabmap]: rtabmap: queue_size      = 30
[rtabmap-1] [INFO] [1713848546.702451929] [rtabmap]: rtabmap: qos_image       = 0
[rtabmap-1] [INFO] [1713848546.702456615] [rtabmap]: rtabmap: qos_camera_info = 0
[rtabmap-1] [INFO] [1713848546.702461347] [rtabmap]: rtabmap: qos_scan        = 0
[rtabmap-1] [INFO] [1713848546.702465832] [rtabmap]: rtabmap: qos_odom        = 0
[rtabmap-1] [INFO] [1713848546.702491715] [rtabmap]: rtabmap: qos_user_data   = 0
[rtabmap-1] [INFO] [1713848546.702496857] [rtabmap]: rtabmap: approx_sync     = true
[rtabmap-1] [INFO] [1713848546.702507500] [rtabmap]: Setup depth callback
[rtabmap-1] [INFO] [1713848546.706323809] [rtabmap]: 
[rtabmap-1] rtabmap subscribed to (approx sync):
[rtabmap-1]    /odom \
[rtabmap-1]    /rgb_camera/image_raw \
[rtabmap-1]    /filtered_depth_image \
[rtabmap-1]    /rgb_camera/camera_info \
[rtabmap-1]    /scan_cloud
[rtabmap-1] [WARN] [1713848562.416830366] [rtabmap]: We received odometry message, but we cannot get the corresponding TF world->camera_depth_optical_frame at data stamp 0.449000s (odom msg stamp is 0.481000s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.
[rtabmap-1] [INFO] [1713848565.885308464] [rtabmap]: rtabmap (1): Rate=0.10s, Limit=0.000s, Conversion=2.9225s, RTAB-Map=0.5285s, Maps update=0.0086s pub=0.0088s (local map=1, WM=1)
[rtabmap-1] [INFO] [1713848566.548250227] [rtabmap]: rtabmap (2): Rate=0.10s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=0.5267s, Maps update=0.0127s pub=0.0264s (local map=2, WM=2)`

It says "We received odometry message, but we cannot get the corresponding TF world->camera_depth_optical_frame at data stamp 0.449000s (odom msg stamp is 0.481000s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once." so noted down the time difference but which parameter I can tweak for adjusting the timestamp delay?

<div ccp_infra_version='3' ccp_infra_timestamp='1713848846105' ccp_infra_user_hash='2122590094' ccp_infra_copy_id='ba0906dd-982e-4068-97b4-ea3a738998d9' data-ccp-timestamp='1713848846105'><html><head><meta name=ProgId content=Excel.Sheet><meta name=Generator content="Microsoft Excel 15"><style>table
	{mso-displayed-decimal-separator:"\.";
	mso-displayed-thousand-separator:"\,";}
tr
	{mso-height-source:auto;}
col
	{mso-width-source:auto;}
td
	{padding-top:1px;
	padding-right:1px;
	padding-left:1px;
	mso-ignore:padding;
	color:black;
	font-size:11.0pt;
	font-weight:400;
	font-style:normal;
	text-decoration:none;
	font-family:"Aptos Narrow", sans-serif;
	mso-font-charset:0;
	text-align:general;
	vertical-align:bottom;
	border:none;
	white-space:nowrap;
	mso-rotate:0;}
.xl18
	{border:.5pt solid black;}
.xl89
	{text-align:center;
	vertical-align:middle;
	border:.5pt solid black;
	white-space:normal;}
.xl90
	{text-align:center;
	vertical-align:middle;
	border:.5pt solid black;}
</style></head><body link="#467886" vlink="#96607D">
Checking tf  at this time | Odom  time | Diff
-- | -- | --
2.242 | 2.282 | 0.04
149.651 | 149.657 | 0.006
0.8 | 0.832 | 0.032
0.438 | 0.471 | 0.033
0.398 | 0.431 | 0.033
0.399 | 0.431 | 0.032
0.623 | 0.656 | 0.033
0.449 | 0.481 | 0.032

</body></html></div>

aasishkc4, 2024-04-23:
These are my parameters "parameters={
          'frame_id':'camera_depth_optical_frame',
        #   'odom_frame_id': 'base_link',
        #   'visual_odometry': False,
          'map_frame_id':'map',
          'approx_sync': True,
          'stereo_approx_sync': True,
          'approx_sync_max_interval': '0.01',
          'subscribe_scan_cloud': True,
          'Grid/Sensor': '2',
        #   'Grid/RayTracing': 'true',
        #   'Grid/3D' : 'true',
          'wait_imu_to_init': True,
          'use_sim_time':use_sim_time,
          'subscribe_depth':True,
        #   'use_action_for_goal':True,
          'map_always_update': True,
        #   'map_cleanup' : True,
          'octomap_tree_depth': 4,
          'cloud_output_voxelized': True,
          'odom_sensor_sync': True,
          'wait_for_transform': 0.5,

        #   'map_empty_ray_tracing': True,
          'tf_delay': 0.05,
          'tf_tolerance': 0.1,
        #   'map_filter_radius' : 5.0,
        #   'map_filter_angle' : 88.0,

        #   'qos_image':qos,
        #   'qos_imu':qos,
        #   'qos_scan':qos,
        #   'qos_odom':qos,
          'queue_size': 30,
          # 'RGBD/LinearUpdate' : '0',
          'Reg/Force3DoF':'true',
          'Optimizer/GravitySigma':'0', # Disable imu constraints (we are already in 2D)
          'Rtabmap/DetectionRate': '10.0'
    }"

matlabbe, 2024-04-28:
That strange, it seems the odometry topic is published before the odom TF topic. That looks more an issue on the odometry plugin used by the simulator. The only downside right now is that pose of the sensors may not be accurate.