introlab/rtabmap #1278: iOS: Memory usage

Peach82, 2024-05-08:
Hello, I'm trying to user RTABMap inside an iOS application but I have some memory problems. 
I added the ViewController in a navigationViewController and it works as expected. 
When I start a new scan the memory usage is about 200MB, the problem is the second scan: when I pop the ViewController with the scanner and then add it back to the navigation stack the initial memory usage is very high (I ended a scan with an M1 iPad Pro at about 1300MB of memory usage, and the next scan started with a memory usage of about 1000MB)
Is there a way to free the memory after popping the RTABMap ViewController from the navigation stack?
Thank You

matlabbe, 2024-05-12:
Unless there is a memory leak, the memory should be reclaimed when starting a new scan. However if on iOS it is like on unix system, the RAM may appear not released though it is. For example, if the second scan is smaller than the first one, the memory usage may remain at 1000 after mapping.

Peach82, 2024-05-15:
Thank You for the reply.
I tried to run the application in Xcode Instruments, and when I pop the scan view controller from the navigation stack I have 9 memory leaks. This is the leaks output, don't know if can be of any help

```
Leaked Object	Count	Address	Size	Responsible Library	Responsible Frame
Malloc 64 Bytes		2	< multiple >	128 Bytes	RTABMapApp	rtabmap::Memory::Memory(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>> const&)
Malloc 32 Bytes		2	< multiple >	64 Bytes	RTABMapApp	cv::aruco::DetectorParameters::create()
Malloc 32 Bytes		2	< multiple >	64 Bytes	RTABMapApp	rtabmap::Rtabmap::parseParameters(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>> const&)
Malloc 32 Bytes		2	< multiple >	64 Bytes	RTABMapApp	cv::Ptr<cv::aruco::Dictionary> cv::makePtr<cv::aruco::Dictionary, cv::aruco::Dictionary>(cv::aruco::Dictionary const&)
Malloc 208 Bytes	2	< multiple >	416 Bytes	RTABMapApp	rtabmap::Rtabmap::parseParameters(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>> const&)
Malloc 176 Bytes	2	< multiple >	352 Bytes	RTABMapApp	cv::aruco::DetectorParameters::create()
Malloc 144 Bytes	2	< multiple >	288 Bytes	RTABMapApp	void std::__1::vector<double, std::__1::allocator<double>>::__assign_with_size[abi:ue170006]<double*, double*>(double*, double*, long)
Malloc 128 Bytes	2	< multiple >	256 Bytes	RTABMapApp	rtabmap::Optimizer::create(rtabmap::Optimizer::Type, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>> const&)
Malloc 112 Bytes	2	< multiple >	224 Bytes	RTABMapApp	cv::Ptr<cv::aruco::Dictionary> cv::makePtr<cv::aruco::Dictionary, cv::aruco::Dictionary>(cv::aruco::Dictionary const&)

```

matlabbe, 2024-05-15:
Total of bytes is around 1.5MB. Most of these functions are called only once, so maybe not the source of the huge memory usage you are seeing. A `cv::Ptr` is a smart pointer, not sure it shows up as a "leak".