introlab/rtabmap #1279: it is possible to increase the frequency map -> icp_odom/odom

qetuo105487900, 2024-05-09:
[image]
based on rtabmap the frequency map -> icp_odom/odom only 20 that too slow

at least, like cartographer the frequency map -> icp_odom/odom only has 195
[image]

qetuo105487900, 2024-05-09:
this is set Rtabmap/DetectionRate as default 1
[image]
the rate only 0.8 hz

this is set Rtabmap/DetectionRate as 30
[image]

the rate only 4.x hz

matlabbe, 2024-05-12:
With `Rtabmap/DetectionRate=30`, if output is ~4Hz, it means rtabmap node requires ~250 ms to process.

for icp_odom -> odom, icp_odometry won't publish faster than input data or could be limited by computer power. If you use a lidar at 10Hz, if icp_odometry is 10 Hz, it means it doesn't drop any frame and is processing as fast as it can.

For map->icp_odom, the default is 20 Hz, but you can make it arbitrary any rate by setting `tf_delay` to `0.005` for 200 Hz (default `0.05` or 20 Hz). See also answer here: https://github.com/introlab/rtabmap/issues/1277#issuecomment-2106397873