introlab/rtabmap #1280: what is the best vio in rtabmap for euroc, or indoor environment?

lturing, 2024-05-10:
fistly, many thanks for your great jobs!
recently, i'm working on vio, and want to try rtabmap, seeking the appropriate or best vio for indoor

matlabbe, 2024-05-13:
I didn't keep track of recent mono-VIO open source approaches since that paper (showing MSCKF-VIO and OKVIS approaches). Not sure if others seeing this post who had experience with some more recent open source VIO approaches could comment here. Recently I've seen some success using [OpenVINS](https://docs.openvins.com/) with rtabmap: https://www.youtube.com/watch?v=WiyaTdKvOmg

Lastly, rtabmap app for android and iOS shows some results using ARCore and ARKit respectively (with ARKit as the best of the two).

You may look at this page for a list of VIO approaches: https://github.com/topics/visual-inertial-odometry?o=desc&s=updated

lturing, 2024-05-13:
thank you very much

closed 2024-05-13 by lturing