introlab/rtabmap #1281: Rtabmap on Android devices

Genozen, 2024-05-15:
Hello, I'm comparing the ios and android version of the Rtabmap and noticed something interesting on the android phones:
[image]

Sometimes the pose jitters a lot. I've searched around the issues and forums but have not seen related mention about this.

Any idea as to why this happens?

matlabbe, 2024-05-15:
Which version are you using? It looks like the pose is jumping between two referentials, there could be a bug. We do offset the origin of the received pose, maybe there is a offset missing sometimes. Let me know which [APK](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-arcore-apk) you are using.

Genozen, 2024-05-20:
@matlabbe  I used the tutorial from here is: https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-arcore-apk.

matlabbe, 2024-05-25:
Reproduced with ARCore-Java driver by moving fast:
[image]

If we move slow it is fine:
[image]

There could be a bug on how arcore localizations are filtered, causing the back-and-forth jumps. In [commit](https://github.com/introlab/rtabmap/commit/c5c190a2a06ca7b1f77cbe45ae6c31f83cb90067) above, I disabled that filtering feature by default. On your version without reinstalling, you can change this setting under Mapping section below:

[image]