introlab/rtabmap #1283: Build error eigen Plugins

adricort, 2024-05-21:
I am trying to build rtabmap and rtabmap_ros from source as indicated in the [instructions](https://github.com/introlab/rtabmap_ros/tree/ros2), but 2 things happen:
1. I get the following error: 

> Starting >>> grid_map_ros
> --- stderr: rtabmap                                                                                                       
> CMake Deprecation Warning at examples/LidarMapping/CMakeLists.txt:1 (cmake_minimum_required):
>   Compatibility with CMake < 2.8.12 will be removed from a future version of
>   CMake.
>
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>
>   CMake that the project does not need compatibility with older versions.

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> compilation terminated.

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> compilation terminated.

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:146: corelib/src/CMakeFiles/rtabmap_core.dir/Memory.cpp.o] Error 1

> gmake[2]: *** Waiting for unfinished jobs....

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:118: corelib/src/CMakeFiles/rtabmap_core.dir/RtabmapThread.cpp.o] Error 1

> compilation terminated.

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:104: corelib/src/CMakeFiles/rtabmap_core.dir/Rtabmap.cpp.o] Error 1

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> compilation terminated.

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:174: corelib/src/CMakeFiles/rtabmap_core.dir/DBDriverSqlite3.cpp.o] Error 1

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> compilation terminated.

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:160: corelib/src/CMakeFiles/rtabmap_core.dir/DBDriver.cpp.o] Error 1

> <command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory

> compilation terminated.

> gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:132: corelib/src/CMakeFiles/rtabmap_core.dir/Statistics.cpp.o] Error 1

> gmake[1]: *** [CMakeFiles/Makefile2:894: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2

> gmake: *** [Makefile:156: all] Error 2

> ---

> Failed   <<< rtabmap [6.81s, exited with code 2]

Is there any modification in the package for it to find the eigen plugins properly? I haven't found so much documentation about it, only [here](https://github.com/ANYbotics/grid_map/issues/67), but I think it is another problem.
2. (maybe not important at the moment) libpointmatcher can't be resolved for the dependencies for jammy, is it required?

matlabbe, 2024-05-26:
The error seems coming from `grid_map_core`. You can disable that dependency with `cmake -DWITH_GRIDMAP=OFF ..`. Thanks for the [link](https://github.com/ANYbotics/grid_map/issues/67), if I can reproduce it I could give a try. What is your machine? Ubuntu 22.04 with ROS2 Humble? From CI, rtabmap should build on jammy/humble even with `ros-humble-grid-map-core` dependency, see log here: https://github.com/introlab/rtabmap/actions/runs/9238128356/job/25415869022

libpointmatcher from `ros-humble-libpointmatcher` should also work.

matlabbe, 2024-10-07:
I tried building RGBDMapping example using rtabmap humble binaries and I don't see the error. 
```
source /opt/ros/humble/setup.bash
cd rtabmap/examples/RGBDMapping
mkdir build
cd build
cmake ..
make VERBOSE=1
```

Here the include directories `make VERBOSE=1`:
`
/usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DDISABLE_PCAP -DDISABLE_PNG -DEIGEN_DENSEBASE_PLUGIN=\"grid_map_core/eigen_plugins/DenseBasePlugin.hpp\" -DEIGEN_FUNCTORS_PLUGIN=\"grid_map_core/eigen_plugins/FunctorsPlugin.hpp\" -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Dkiss_fft_scalar=double -I/home/mathieu/workspace/rtabmap/examples/RGBDMapping/build/rgbd_mapping_autogen/include -isystem /opt/ros/humble/include/rtabmap-0.21 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/x86_64-linux-gnu/qt5/QtOpenGL -isystem /usr/include/opencv4 -isystem /usr/include/pcl-1.12 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /opt/ros/humble/include -isystem /usr/include/openni2 -isystem /usr/include/vtk-9.1 -isystem /usr/include/jsoncpp -isystem /usr/include/freetype2 -fPIC -MD -MT CMakeFiles/rgbd_mapping.dir/main.cpp.o -MF CMakeFiles/rgbd_mapping.dir/main.cpp.o.d -o CMakeFiles/rgbd_mapping.dir/main.cpp.o -c /home/mathieu/workspace/rtabmap/examples/RGBDMapping/main.cpp
`

I see `-isystem /opt/ros/humble/include` and on my system, the files `grid_map_core/eigen_plugins/DenseBasePlugin.hpp` and `grid_map_core/eigen_plugins/FunctorsPlugin.hpp` are indeed located under `/opt/ros/humble/include`:
```
$ ls /opt/ros/humble/include/grid_map_core/eigen_plugins/*
/opt/ros/humble/include/grid_map_core/eigen_plugins/DenseBasePlugin.hpp
/opt/ros/humble/include/grid_map_core/eigen_plugins/Functors.hpp
/opt/ros/humble/include/grid_map_core/eigen_plugins/FunctorsPlugin.hpp
```
Are the include directories different on your system?

matlabbe, 2024-10-09:
I [removed](https://github.com/introlab/rtabmap/commit/465ea85d91d8340ccc961f5997333e7e16c8258a) grid_map from the public interface. That change will be pushed on next ros1/ros2 binary release.

parkerlreed, 2025-02-27:
I am seeing this with the current rtabmap-jammy container and up to date humble from apt

Made sure to clean my build folder 

```
<command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory
compilation terminated.
<command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory
compilation terminated.
make[2]: *** [tools/Info/CMakeFiles/info.dir/build.make:76: tools/Info/CMakeFiles/info.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1356: tools/Info/CMakeFiles/info.dir/all] Error 2
make[2]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/build.make:76: tools/StereoEval/CMakeFiles/stereoEval.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1112: tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
<command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory
compilation terminated.
make[2]: *** [tools/Export/CMakeFiles/export.dir/build.make:76: tools/Export/CMakeFiles/export.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1301: tools/Export/CMakeFiles/export.dir/all] Error 2
<command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory
compilation terminated.
make[2]: *** [tools/DetectMoreLoopClosures/CMakeFiles/detectMoreLoopClosures.dir/build.make:76: tools/DetectMoreLoopClosures/CMakeFiles/detectMoreLoopClosures.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1274: tools/DetectMoreLoopClosures/CMakeFiles/detectMoreLoopClosures.dir/all] Error 2
[ 62%] Linking CXX executable ../../bin/rtabmap-recovery
[ 63%] Linking CXX executable ../../bin/rtabmap-imagesJoiner
[ 63%] Built target recovery
[ 63%] Built target imagesJoiner
[ 63%] Built target rtabmap_gui_autogen
make: *** [Makefile:156: all] Error 2
```

matlabbe, 2025-02-27:
If you build rtabmap-jammy, make sure you have already the latest `introlab3it/rtabmap-jammy-deps` image downloaded.

Not sure what you mean by installing apt packages, but if you rebuild rtabmap from source inside the container, make sure grid_map_core package is not installed.

parkerlreed, 2025-02-28:
Yep that was it, thanks

```
ii  ros-humble-grid-map-core                          2.0.1-1jammy.20250121.123922            amd64        Universal grid map library to manage two-dimensional grid maps with multiple data layers.
```