introlab/rtabmap #1288: Docker container libs updated without rtabmap rebuild

parkerlreed, 2024-05-25:
Bunch of library updates on the humble jammy container. 

rtabmap was not rebuilt against the updates (librealsense2 in this case)

```
Setting up ros-humble-ros-base (0.10.0-1jammy.20240522.174356) ...
Setting up ros-humble-nav2-bringup (1.1.14-1jammy.20240518.064327) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.7) ...
[parker@rogally ~]$ .local/bin/rtabmap 
/usr/local/bin/rtabmap: error while loading shared libraries: librealsense2.so.2.54: cannot open shared object file: No such file or directory```

matlabbe, 2024-05-27:
Which docker image exactly?

Just tried this one and it seems to work (can launch D435i and see data), following [this](https://github.com/introlab/rtabmap/wiki/Installation#docker):
```
docker pull introlab3it/rtabmap:22.04

export XAUTH=/tmp/.docker.xauth
touch $XAUTH
xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -

 docker run -it --rm \
   --privileged \
   --env="DISPLAY=$DISPLAY" \
   --env="QT_X11_NO_MITSHM=1" \
   --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
   --env="XAUTHORITY=$XAUTH" \
   --volume="$XAUTH:$XAUTH" \
   -e NVIDIA_VISIBLE_DEVICES=all \
   -e NVIDIA_DRIVER_CAPABILITIES=all \
   --runtime=nvidia \
   --network host \
   -v ~/Documents/RTAB-Map:/root/Documents/RTAB-Map \
   introlab3it/rtabmap:22.04 \
   rtabmap
```

Oh, do you mean if you update libraries inside the container it will fail? I could force trigger the rebuild of the base image, it is maybe using an old cache for a while.

parkerlreed, 2024-05-27:
Yeah the humble jammy I ran a manual apt-update which updated the base libs but not rtabmap itself, so it didn't get a server side rebuild.

parkerlreed, 2024-05-31:
For Docker I can obviously sit on the base image without updates, but I assume this affects anyone with Ubuntu 22.04 on bare metal with the ROS repos as well. 

librealsense2 updated to 2.55 and rtabmap just needs a rebuild against it.

matlabbe, 2024-06-02:
The [introlab3it/rtabmap:jammy](https://hub.docker.com/layers/introlab3it/rtabmap/jammy/images/sha256-e25e24d124f45445901f27b67949c432b23a0796c0d5a9cbf44be9310b202fc9) has been updated, it should use 2.55 now:
```
docker run -it --rm introlab3it/rtabmap:jammy ldd /usr/local/bin/rtabmap | grep realsense
	librealsense2.so.2.55 => /opt/ros/humble/lib/x86_64-linux-gnu/librealsense2.so.2.55 (0x00007138e3073000)
```

parkerlreed, 2024-06-03:
Just apt updated the container and still seeing the older rtabmap. Was the base image itself updated?

```
[parker@rogally ~]$ sudo apt update
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:3 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
[parker@rogally ~]$ rtabmap
rtabmap: error while loading shared libraries: librealsense2.so.2.54: cannot open shared object file: No such file or directory
```

parkerlreed, 2024-06-03:
Aha! I see it was. Thanks. Works after a fresh image pull.

closed 2024-06-03 by parkerlreed