introlab/rtabmap #1295: No color images, only grayscale images

alphabet-lgtm, 2024-06-03:
As shown in the figure below, I am using a d435i camera but cannot obtain a color RGB image
[image]

alphabet-lgtm, 2024-06-04:
> Is that on standalone? You can change the parameters there to use RGB-D mode instead: ![image](https://private-user-images.githubusercontent.com/2319645/336171768-d57e9471-4a18-4350-9770-877a3128da71.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0NjM2NjgsIm5iZiI6MTcxNzQ2MzM2OCwicGF0aCI6Ii8yMzE5NjQ1LzMzNjE3MTc2OC1kNTdlOTQ3MS00YTE4LTQzNTAtOTc3MC04NzdhMzEyOGRhNzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDRUMDEwOTI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGY3NzFlZjYyNjAyNTcyMThhZmZjMDcwYTE1MDk4Yjk1MjI2YzBkZGNlM2JiZjEzY2Y3MTY2YTFlNjI2MmQyZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Qq2q7LCj0TgQCTGM0AWTJWiPZraX5Ll_OAN5pZOHI_M)
> 
> **However**, on D435i the odometry accuracy will be worst because the RGB camera has smaller FOV and has a rolling shutter (not totally sure if RGB camera is perfectly in sync with ir cameras too). On D455, it can be okay as they use a different RGB camera (with global shutter), but never experienced myself.

Thank you for your response. I successfully obtained the color image. Regarding odometry accuracy, I don't have any other cameras and have no way to verify it.

closed 2024-06-04 by alphabet-lgtm

matlabbe, 2024-06-04:
In RGB-D mode, you may see the camera jumping more and drifting more when you make a loop.

alphabet-lgtm, 2024-06-04:
> In RGB-D mode, you may see the camera jumping more and drifting more when you make a loop.

Yes, I have found out about it, but I just need color image even there will be some problems.