introlab/rtabmap #1304: GUI is blank

alphabet-lgtm, 2024-06-21:
[image]
[image]

GUI is blank but may still work. I have tried [RTABMap-0.21.4-win64-cuda117.7z] and [RTABMap-0.21.4-win64-cuda117.exe] but got the same problem.

matlabbe, 2024-06-24:
Can I assume the first example was the version without cuda from this link? https://github.com/introlab/rtabmap/releases/tag/0.21.4

I cannot see that problem on my Windows computer (Windows 11 Home). Which Windows version? Is that a computer with a dedicated or integrated GPU (though it should not matter)?

The only way to reproduce a similar issue is to close all panels from the main window:
[image]
though the menu bar should still be there to reopen all panels (under Window menu).

alphabet-lgtm, 2024-06-24:
> Can I assume the first example was the version without cuda from this link? https://github.com/introlab/rtabmap/releases/tag/0.21.4
>
> I cannot see that problem on my Windows computer (Windows 11 [...]

The version with cuda from the link(https://github.com/introlab/rtabmap/releases/tag/0.21.4).The version without cuda can work properly, but I want to use it with cuda. I used win10 with cuda11.7. Does this warning affect?

[image]

[image]

matlabbe, 2024-06-24:
For the PATH error, it can be ignored if you use just the app (not using rtabmap as a library). I cannot reproduce the issue with the cuda version. You may try older versions like 0.21.0: https://github.com/introlab/rtabmap/releases

CUDA version won't improve the general speed of VSLAM. Use it only if you really need a feature that is provided only with CUDA build.

alphabet-lgtm, 2024-06-24:
I have tried [v0.21.0](https://github.com/introlab/rtabmap/releases/tag/0.21.0) and it just stuck and flashback. 
Next, I tried [v0.20.16](https://github.com/introlab/rtabmap/releases/tag/0.20.16) and it also stuck like this.
[image]

I want to improve the speed of Loop closure dection because it worked so slowly(few seconds one frame).