introlab/rtabmap #1306: Regression on RHEL for Jazzy with latest 0.21.5 version

marcoag, 2024-06-26:
With the [latest bump to 0.21.5-1](https://github.com/ros/rosdistro/commit/73a3da4fd32416fcfec662ffd7b04866a7a5b4f1) `rtabmap` seems to be hitting a regression for [RHEL in Jazzy Jalisco](http://repo.ros2.org/status_page/ros_jazzy_rhel.html?q=REGRESSION). This regression will remove the package in the upcoming Jazzy sync (coming soon) from the `RHEL` repos.

To avoid this I will proceed to downgrade to latest working version on the rosdistro `0.21.1-4`. Since we don't have a per platform versioning system this means all supported OS will get the downgraded version. This will give you some time to figure out the regression while keeping the package available to users

Let me know your thoughts or if wanna proceed in a different way.

matlabbe, 2024-06-26:
Alright, you can revert for now.

About the issue, the build error seems coming from upstream package `ros-jazzy-gtsam`:
```
06:38:20 DEBUG: [ 31%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o
06:38:20 DEBUG: cd /builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/.obj-x86_64-redhat-linux-gnu/corelib/src && /usr/bin/g++ -DDISABLE_LIBUSB -DDISABLE_PCAP -DDISABLE_PNG -DPCL_OMP -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DVTK_GLOBAL_WARNING_DISPLAY_OFF -Dkiss_fft_scalar=double -Dqh_QHpointer -Drtabmap_core_EXPORTS -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/corelib/src -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/corelib/src/../include -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/.obj-x86_64-redhat-linux-gnu/corelib/src -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/.obj-x86_64-redhat-linux-gnu/corelib/src/include -I/opt/ros/jazzy/lib/cmake/GTSAM/../../../include -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/utilite/src/../include -I/builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/.obj-x86_64-redhat-linux-gnu/utilite/src/include -isystem /opt/ros/jazzy/include -isystem /usr/include/eigen3 -isystem /usr/include/opencv4 -isystem /usr/include/pcl-1.12 -isystem /usr/include/ni -isystem /usr/include/vtk -isystem /usr/include/freetype2 -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtOpenGL -isystem /opt/ros/jazzy/include/g2o/core -isystem /opt/ros/jazzy/include/g2o/stuff -isystem /opt/ros/jazzy/include/g2o/solvers/eigen -isystem /opt/ros/jazzy/include/g2o/solvers/pcg -isystem /opt/ros/jazzy/include/g2o/types/slam2d -isystem /opt/ros/jazzy/include/g2o/types/slam3d -isystem /opt/ros/jazzy/include/g2o/types/sba -isystem /opt/ros/jazzy/include/g2o/solvers/csparse -isystem /usr/include/suitesparse -isystem /opt/ros/jazzy/include/g2o/solvers/cholmod -isystem /opt/ros/jazzy/include/gtsam/3rdparty/SuiteSparse_config -isystem /opt/ros/jazzy/include/gtsam/3rdparty/CCOLAMD -isystem /opt/ros/jazzy/include/gtsam/3rdparty/metis -isystem /opt/ros/jazzy/include/gtsam/3rdparty/Eigen -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64 -march=x86-64-v2 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fmessage-length=0 -fopenmp -std=c++17 -DNDEBUG -std=gnu++17 -fPIC   -Wall -Wno-unknown-pragmas -fPIC -fopenmp -MD -MT corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o -MF CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o.d -o CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o -c /builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/corelib/src/optimizer/OptimizerGTSAM.cpp
06:38:23 DEBUG: In file included from /opt/ros/jazzy/include/gtsam/global_includes.h:22,
06:38:23 DEBUG:                  from /opt/ros/jazzy/include/gtsam/base/Vector.h:28,
06:38:23 DEBUG:                  from /opt/ros/jazzy/include/gtsam/base/Matrix.h:28,
06:38:23 DEBUG:                  from /opt/ros/jazzy/include/gtsam/base/Manifold.h:22,
06:38:23 DEBUG:                  from /opt/ros/jazzy/include/gtsam/geometry/BearingRange.h:21,
06:38:23 DEBUG:                  from /opt/ros/jazzy/include/gtsam/geometry/Pose2.h:23,
06:38:23 DEBUG:                  from /builddir/build/BUILD/ros-jazzy-rtabmap-0.21.5/corelib/src/optimizer/OptimizerGTSAM.cpp:39:
06:38:23 DEBUG: /opt/ros/jazzy/include/gtsam/base/types.h:34:10: fatal error: tbb/scalable_allocator.h: No such file or directory
06:38:23 DEBUG:    34 | #include <tbb/scalable_allocator.h>
06:38:23 DEBUG:       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
06:38:23 DEBUG: compilation terminated.
06:38:23 DEBUG: make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:947: corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerGTSAM.cpp.o] Error 1
```

The 0.21.1 release didn't depend on gtsam ros package, which is why it was building previously.

matlabbe, 2024-06-27:
It looks like reverting to old version made the ubuntu noble fail: https://build.ros2.org/job/Jbin_uN64__rtabmap__ubuntu_noble_amd64__binary/ for a different reason.

It seems the RHEL issue with GTSAM won't be fixed soon, I don't have anyway to test on RHEL (I don't have a subscription) and that I really want to keep gtsam dependency for Jazzy on Noble, what if we simply blacklisting rtabmap for RHEL? It seems that it have been done in the past for rtabmap on Humble and rhel8 but I don't know who did it or where to do it: https://build.ros2.org/job/Hbin_rhel_el864__rtabmap__rhel_8_x86_64__binary/

> Generated at 2024-03-21 22:34:38 -0800 from template 'release/rpm/binarypkg_job.xml.em' but disabled since the package is blacklisted (or not whitelisted) in the configuration file
This project is currently disabled 

We could reinstate the latest version working on ubuntu noble while disabling rtabmap for rhel9.

marcoag, 2024-06-27:
Oh, that doesn't seem good. Let me have a look at the options regarding blacklisting for RHEL.

Btw, if you want to test your RHEL builds you can use [Alma](https://almalinux.org/get-almalinux/) or [Rocky](https://rockylinux.org/download), no license required.

matlabbe, 2024-06-27:
Thanks for the tip, I could reproduce the GTSAM issue with a simple Dockerfile. See https://github.com/borglab/gtsam/issues/1769

I suggested them a fix, you may comment if it makes sense.

marcoag, 2024-06-28:
Nice, looking good. Thanks for the fast fix. Since there's an upstream fix on the way I might not eve blacklist this and just revert to the noble working version.

matlabbe, 2024-07-01:
To avoid waiting for next gtsam release, I just removed gtsam dependency in that new release https://github.com/ros/rosdistro/pull/41892 for jazzy.

marcoag, 2024-07-18:
@matlabbe the latest `jazzy` version of rtabmap seemed to be failing for `noble` so I rolled it back so at least it doesn't get deleted on the next sync: https://github.com/ros/rosdistro/pull/42124