introlab/rtabmap #1308: RPATH needs to be set for dynamic linking

Serafadam, 2024-07-05:
When trying to link against shared lib targets (for example rtabmap::utilite) one needs to set `--disable-new-dtags` for target GCC linker options or otherwise shared libs cannot be found at runtime (rtabmap::core works normally, but rtabmap::utilite cannot be found even though it is in the same directory). This also causes some issues when trying to build on MacOS/Clang setup.

matlabbe, 2024-07-21:
Is this issue only on Mac OSX? On linux, I have included in another project the library from the build directory (not installed) and it worked. When launching the third party app, we should still add in the PATH the bin directory though to be executed.

Serafadam, 2024-07-29:
This also happened on Linux, currently there have been some changes but I can point to some temporary branch that still has those parts of code. [Here](https://github.com/luxonis/depthai-core/blob/b01d409668b632aef7dcaeba219a01259fd2f7f7/CMakeLists.txt#L754) we set up a target that links to RTABMap's libraries and can be used to link to our examples. But if we want to create the example as in [here](https://github.com/luxonis/depthai-core/blob/b01d409668b632aef7dcaeba219a01259fd2f7f7/examples/CMakeLists.txt#L477), then everything build correctly but linker cannot find rtabmap::utilite target (this can be investigated with ld), even though rtabmap_utilite.so is in the same directory as rtabmap::core which is found properly. With `--disable-new-dtags` it works properly but it seems not all compilers have this option.

matlabbe, 2024-08-02:
I'll try to create a minimal example on my side to reproduce the problem, based on your examples. I won't be able to work on this the next 2 weeks. I'll keep you updated afterwards.

matlabbe, 2024-08-26:
I tried to dig more in the cmake/linker lore to know what is happening. That tag `--disable-new-dtags` is used to disable usage of `RUNPATH` and use old `RPATH` on recent compilers. Here is a [video](https://youtu.be/m0DwB4OvDXk?si=x8mdocmH66DqXosN) (found from this thread [Setting RPATH and not RUNPATH for executable](https://discourse.cmake.org/t/setting-rpath-and-not-runpath-for-executable/10011)) explaining the difference between `RPATH` and `RUNPATH` (check from 45:10). Basically, `LD_LIBRARY_PATH` is looked after `RPATH` in old behavior but before `RUNPATH` in new behavior. Another related post: https://stackoverflow.com/questions/52018092/how-to-set-rpath-and-runpath-with-gcc-ld

That explains why on Ubuntu 22.04, I started to see `rtabmap` launched from build directory loading shared rtabmap's library installed in ROS binaries directory `/opt/ros/humble/lib/...` instead of those built in same directory than the executable. I thought it was a new behavior of recent OS, but actually it can use libraries built in same directory, it was just seeing the same ones installed by ROS because ROS adds them to the `LD_LIBRARY_PATH`. So ideally, keep `LD_LIBRARY_PATH` empty unless you really want to use specific libraries, and `RUNPATH` should then find the correct libraries from the build directory. Because ROS seems to update `LD_LIBRARY_PATH` anyway, a workaround when developping in build directories is to add `::` at the begging of `LD_LIBRARY_PATH` so that current directory of where you launch an executable is looked first:
```
echo "export LD_LIBRARY_PATH=::$LD_LIBRARY_PATH" >> ~/.bashrc
```

For reference:
 * Command to see if `RPATH` or `RUNPATH` is used in an executable, and check if the path is good (here interestingly I was linking to an OpenCV version built from source and installed in its own build directory): 
    ```bash
    $ cd ~/workspace/rtabmap/build/bin
    $ readelf -d ./rtabmap |head -20
    [...]
     0x000000000000001d (RUNPATH) Library runpath: [/home/mathieu/workspace/rtabmap/build/bin:/home/mathieu/workspace/opencv/build/install/lib:/opt/ros/humble/lib/x86_64-linux-gnu:]
    [...]
    ```

**In conclusion**, as `RUNPATH` seems the default new approach, I would not set `--disable-new-dtags` by default. So I added in this commit https://github.com/introlab/rtabmap/commit/f73e9c02b3b7d624c454b2ab096e11ce6171a972 a new cmake option called `BUILD_WITH_RPATH_NOT_RUNPATH` that you can activate for your internal build:
```bash
$ cd ~/workspace/rtabmap/build/
$ cmake -DBUILD_WITH_RPATH_NOT_RUNPATH=ON ..
[...]
-- --------------------------------------------
-- Info :
--   RTAB-Map Version =     0.21.6
--   CMAKE_VERSION =        3.22.1
--   CMAKE_INSTALL_PREFIX = /home/mathieu/workspace/rtabmap/build_cv_latest/install
--   CMAKE_BUILD_TYPE =     Release
--   CMAKE_INSTALL_LIBDIR = lib
--   BUILD_APP =            ON
--   BUILD_TOOLS =          ON
--   BUILD_EXAMPLES =       ON
--   BUILD_SHARED_LIBS =    ON
--   CMAKE_CXX_FLAGS =  -fmessage-length=0 -fopenmp -std=c++17
--   LINKER FLAGS =    --disable-new-dtags                      <-------------------- new
[...]
$ make
$ cd bin
$ readelf -d ./rtabmap |head -20
[...]
 0x000000000000000f (RPATH)  Library rpath: [/home/mathieu/workspace/rtabmap/build/bin:/home/mathieu/workspace/opencv/build/install/lib:/opt/ros/humble/lib/x86_64-linux-gnu:]
[...]
```

cheer,
Mathieu

closed 2024-08-26 by matlabbe