introlab/rtabmap #1309: Using Fast GP3 and Organized Surface Reconstruction via CLI

nirmalsnair, 2024-07-08:
Hi @matlabbe,
Is it possible to use the `Fast GP3` or `Organized` surface reconstruction methods instead of `Poisson` via command line?
In my tests, I found `Poisson` (55s) to be much slower compared to `Fast GP3` (20s) and `Organized` (5s).

matlabbe, 2024-07-21:
You would have to change the code here:
https://github.com/introlab/rtabmap/blob/08a44ec44f2cc9c68c64bd3e7a75bbcaeb71fa38/tools/Export/main.cpp#L1798-L1805
and change it for gp3 following that tutorial for example: https://pcl.readthedocs.io/projects/tutorials/en/latest/greedy_projection.html#greedy-triangulation

For `Organized`, that would be more difficult to do, but not sure this is an useful reconstruction though. `Organized` creates meshes independently for each image, it doesn't create mesh on the merged cloud like Poisson and GP3.

To increase speed of Poisson, you can lower the resolution with `--poisson_depth` (set to 8 or 9)

nirmalsnair, 2024-07-23:
Thanks @matlabbe, will try to modify the code to make `Fast GP3` available via CLI.

nirmalsnair, 2024-07-23:
> Organized creates meshes independently for each image, it doesn't create mesh on the merged cloud like Poisson and GP3.

On my test data, `Organized` is generating a single mesh just like the other two methods.

#### (Generated via GUI using the default export parameter values with 'Meshing' and 'Texture mapping' enabled)

`Poisson`
[image]

`Fast GP3`
[image]

`Organized`
[image]

matlabbe, 2024-07-23:
Organized generates one file, but some polygons taken from different cameras may overlap. For example, when two cameras overlap, you would see two surfaces overlapping when zooming in. With Poisson, only one surface would be generated (like an average of the two surfaces).

nirmalsnair, 2024-07-24:
Understood, thanks for the clarification.

On a different note, I've noticed that during the mesh export, the GPU usage remains almost zero, even when using the RTAB-Map CUDA version ([RTABMap-0.21.4-win64-cuda117.exe](https://github.com/introlab/rtabmap/releases/download/0.21.4/RTABMap-0.21.4-win64-cuda117.exe)). Is this typical? Is there a way to utilize the GPU to speed up the mesh export?

matlabbe, 2024-07-24:
The cuda version doesn't convert the whole code to cuda, only some parts of it and there is always an explicit parameter attached to (e.g. some visual features can be computed on gpu). If you use a ZED camera, it is better to use cuda version (rectification/depth computed on gpu).

nirmalsnair, 2024-07-25:
We are using the iOS app to capture data without any issues. On the PC, we only perform mesh export and are curious why the GPU is not utilized at all.

matlabbe, 2024-07-27:
Under the hood, we use PCL mesh reconstruction, which is running on CPU.

closed 2024-08-12 by nirmalsnair