introlab/rtabmap #1319: build issue

luke0911, 2024-08-01:
I downloaded the source and tried to build it on my iPhone 12, but I got a build error. At first, there was an rtabmap/utilite/ULogger.h' file not found error, and I tried to fix it through header search paths. Of course, I also completed installing the command through terminal. The header search paths changed and proceeded as follows, but another error occurred. Please check. I think I've done all the related methods.
The environment is as follows.
macOS Sonoma 14.6
iPhone 12 17.5.1
Xcode 15.0.1
[image]

matlabbe, 2024-08-02:
See "To build from source" guide here: https://github.com/introlab/rtabmap/wiki/Installation#ios

drawrs, 2024-09-03:
> https://github.com/introlab/rtabmap/wiki/Installation#ios

Hi, got another error after adding Search Path, I need your help to fix this. Thanks
[image]

drawrs, 2024-09-03:
I did follow the guide. I'm using cmake version 3.30.3

matlabbe, 2024-09-06:
Can you find that file (`utilite_export.h`) under `rtabmap/app/ios/RTABMapApp/Libraries/include/rtabmap-0.21/rtabmap/utilite` ?

Note that if `rtabmap/app/ios/RTABMapApp/Libraries/include/rtabmap-0.21` folder is missing, something went wrong when executing script [install_deps.sh](https://github.com/introlab/rtabmap/blob/master/app/ios/RTABMapApp/install_deps.sh).