introlab/rtabmap #1320: Add very low RAM mode for iOS App large-scale scanning

matlabbe, 2024-08-02:
Like android app's "Trajectory Mode", we could do something similar for iOS. This could save battery and make it able to scan in order of a hour or more. See this thread for more info: http://official-rtab-map-forum.206.s1.nabble.com/iOS-best-practices-for-only-tracking-long-recordings-td10300.html

zeeyang, 2024-10-09:
Is it possible to post process the recorded data on device to generate optimized poses? And what's the equivalent in CLI? Is it `rtab-reprocess`?

matlabbe, 2024-10-10:
I didn't implement a way to reprocess on device. 

The GUI way to reprocess is to open RTAB-Map Desktop, open Preferences->Source, then select Database as input source. Scroll down to select the path of the recorded database. Say yes to use odometry from the database, but no for setting rtabmap detection rate to 0. Check database stamps for framerate. Go back on main UI and press play.

To use the CLI tool:
```
rtabmap-reprocess \
   --Mem/RehearsalSimilarity 0.3 \
   --Kp/MaxFeatures 400 \
   --Rtabmap/MaxRetrieved 2\
   --RGBD/MaxLocalRetrieved  2\
   --Mem/MapLabelsAdded true \
   --Rtabmap/MemoryThr 0 \
   --Mem/STMSize 10 \
   --RGBD/ProximityBySpace true \
   --RGBD/LinearUpdate 0.05 \
   --RGBD/AngularUpdate 0.05 \
    recording.db output.db
```

matlabbe, 2024-10-13:
Closing as already available on iOS app under Menu -> Advanced -> New Data Recording.

closed 2024-10-13 by matlabbe