introlab/rtabmap #1321: Getting the current camera pose

bingogome, 2024-08-06:
I apologize if this is a dumb question :)

I use rtabmap->getLastLocalizationPose() to get a pose but the update rate seems pretty low and sometimes it does not get updated. I am not using ROS so tf is out of question. I am following the tutorial [here](https://github.com/introlab/rtabmap/blob/master/examples/RGBDMapping/main.cpp) so I suppose it is in the mapping mode. What should be the proper way?

matlabbe, 2024-08-12:
Here: 
https://github.com/introlab/rtabmap/blob/4911dbe9bbfcdaebc1a60bc2a72c5cdc5f614bd6/examples/RGBDMapping/MapBuilder.h#L158

We transform odometry pose in map frame (using last map->odom correction received) at odometry frame rate. `getLastLocalizationPose()` is updated only when there is a loop closure in mapping mode, so probably not what you want.

bingogome, 2024-08-16:
@matlabbe Thank you so much for your reply! It now works great. Closing the issue

closed 2024-08-16 by bingogome