introlab/rtabmap #1322: Please help me.. there are too many obstacles in the graph view.

kuacboss, 2024-08-07:
[image]
I am a beginner. I mapped the school hallway and it looks like this even though there are no obstacles. Does this black color represent matched feature points? If so, is there a way to prevent me from recognizing the ceiling or floor features as obstacles (black)?

matlabbe, 2024-08-12:
you can look at `Grid/` parameters:
```
rtabmap --params | grep Grid/
[...]
Param: Grid/MaxObstacleHeight = "0.0"                      [Maximum obstacles height (0=disabled).]
```

By default the floor should not be detected as obstacle. How are you mapping? in 2D, 3D? on a robot?

kuacboss, 2024-08-14:
I am using realsense 455 for my robot. The biggest problem is that, as in other maps, when you visit the same place on a 2D map, the previous map information is erased and a black point cloud appears, ruining the map.
[image]
[image]

matlabbe, 2024-08-15:
Can you share the database?

If stereo data gets too noisy, you may just use a simple thresholding of the floor by turning off normals segmentation (`Grid/NormalsSegmentation=false`) and set `Grid/MaxGroundHeight` to `0.1` m for example.

Glen02lee, 2025-07-21:
안녕하세요.
저도 realsense d455를 사용하고 있는 사람인데 저도 똑같은 문제가 생기고 있습니다.
저는 다음과 같은 스크립트를 사용하고 있습니다.

천장이 너무 많이검출되어 rangemax를 줄이는 방법을 쓰고 있는데 좋지 않은 방법인것 같습니다.

 
#!/bin/bash

echo "🟢 [1/6] Launching RealSense D455 driver..."
gnome-terminal --tab -- bash -c "ros2 launch realsense2_camera rs_launch.py \
  camera_name:=d455 \
  device_type:=d455 \
  depth_module.depth_profile:=640x480x30 \
  rgb_camera.color_profile:=640x480x30 \
  enable_gyro:=true \
  enable_accel:=true \
  unite_imu_method:=1 \
  enable_sync:=true \
  align_depth.enable:=true \
  publish_tf:=true \
  tf_publish_rate:=30.0; exec bash"

sleep 5

echo "🟢 [2/6] Launching IMU filter (Madgwick)..."
gnome-terminal --tab -- bash -c "ros2 run imu_filter_madgwick imu_filter_madgwick_node \
  --ros-args -p use_mag:=false -p publish_tf:=false -p world_frame:=enu \
  -r /imu/data_raw:=/camera/d455/imu -r /imu/data:=/rtabmap/imu; exec bash"

sleep 2

echo "🟢 [3/6] Publishing TF: base_link -> d455_link..."
gnome-terminal --tab -- bash -c "ros2 run tf2_ros static_transform_publisher \
  0.09 0 0.57 0 0 0 base_link d455_link; exec bash"

sleep 3

echo "🟢 [4/6] Launching RTAB-Map in SLAM mode with fresh DB..."
gnome-terminal --tab -- bash -c "source ~/rtab_ws/install/setup.bash && \
ros2 launch rtabmap_launch rtabmap.launch.py \
  rtabmap_args:=\"--delete_db_on_start --Grid/RangeMin 0.3 --Grid/RangeMax 2.5 --Grid/MaxGroundHeight 0.0\" \
  rgb_topic:=/camera/d455/color/image_raw \
  depth_topic:=/camera/d455/aligned_depth_to_color/image_raw \
  camera_info_topic:=/camera/d455/aligned_depth_to_color/camera_info \
  subscribe_depth:=true \
  subscribe_rgb:=true \
  frame_id:=base_link \
  queue_size:=30 \
  wait_imu_to_init:=false \
  rviz:=true; exec bash"

echo "✅ SLAM with new DB launched successfully!"