introlab/rtabmap #1327: Docker images - unable to choose libfreenect for kinekt xbox360

kallibr44, 2024-08-12:
what should i do for enabling this option? Tried docker image from instruction
[image]

matlabbe, 2024-08-15:
Freenect would need to be installed (`apt install libfreenect-dev`) in the docker image and rtabmap rebuilt. Which docker image are you using? I think that dependency was there before but after some refactoring of rtabmap_ros, it was not installed anymore. I could check to re-add it starting with the docker image you are using.

kallibr44, 2024-08-15:
> Freenect would need to be installed (`apt install libfreenect-dev`) in the docker image and rtabmap rebuilt. Which docker image are you using? I think that dependency was there before but after some [...]

yeah, i researched all sources for docker images, at least in focal image (also tried jammy, but based image has trouble with apt module), it have freenect2 library, which used for newer kinect module, but not old. Rebuilt with first freenect lib and it's worked. Successfully tested at Steam Deck :)

closed 2024-08-15 by kallibr44

matlabbe, 2024-08-19:
FYI I added `libfreenect-dev` dependency to docker images: https://github.com/introlab/rtabmap/commit/6ce85b4e55adc2903a6f08dd25fa0c894b34c9cd