introlab/rtabmap #1332: Performance difference on Windows and Ubuntu?

bingogome, 2024-08-27:
Hi,

Thank you for your amazing work!

I noticed some performance difference on Windows and Ubuntu while I move the camera. It seems on Windows the perform is better. Is this somewhat expected?  On Windows, I used the released installer and on Ubuntu built from the source. I wonder if there is a way to look at the dependency, setting difference etc? 

Also, the Odometry Viewer on Ubuntu has some disruption. It doesn't behave the same as in the main app. 

Some screen recordings: 
https://drive.google.com/file/d/1UQEgPbYgSD34_7R4hAmkF4EBAZ82TDMm/view?usp=drive_link, 
https://drive.google.com/file/d/1Dv-Cst6LjcdB3bPJlUVlBjN3m-Bz9khN/view?usp=drive_link, https://drive.google.com/file/d/15PEXB750kSI2SmcGjfcsslEUMcxCQ8aU/view?usp=drive_link

My CMake:
```
 ANDROID_PREBUILD                 OFF
 BUILD_APP                        ON
 BUILD_EXAMPLES                   ON
 BUILD_SHARED_LIBS                ON
 BUILD_TOOLS                      ON
 BUILD_WITH_RPATH_NOT_RUNPATH     OFF
 Boost_DATE_TIME_LIBRARY_RELEAS   /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.74.0
 Boost_FILESYSTEM_LIBRARY_RELEA   /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0
 Boost_INCLUDE_DIR                /usr/include
 Boost_IOSTREAMS_LIBRARY_RELEAS   /usr/lib/x86_64-linux-gnu/libboost_iostreams.so.1.74.0
 Boost_SERIALIZATION_LIBRARY_RE   /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0
 Boost_SYSTEM_LIBRARY_RELEASE     /usr/lib/x86_64-linux-gnu/libboost_system.so.1.74.0
 CHOLMOD_INCLUDE_DIR              CHOLMOD_INCLUDE_DIR-NOTFOUND
 CHOLMOD_LIB                      CHOLMOD_LIB-NOTFOUND
 CMAKE_BUILD_TYPE
 CMAKE_INSTALL_PREFIX             /usr/local
 CSPARSE_INCLUDE_DIR              CSPARSE_INCLUDE_DIR-NOTFOUND
 CSPARSE_LIBRARY                  CSPARSE_LIBRARY-NOTFOUND
 DC1394_INCLUDE_DIRS              /usr/include/dc1394
 DC1394_LIBRARIES                 /usr/lib/x86_64-linux-gnu/libdc1394.so
 FLANN_INCLUDE_DIR                /usr/include
 FLANN_KDTREE_MEM_OPT             OFF
 FLANN_LIBRARY_DEBUG_SHARED       FLANN_LIBRARY_DEBUG_SHARED-NOTFOUND
 FLANN_LIBRARY_DEBUG_STATIC       FLANN_LIBRARY_DEBUG_STATIC-NOTFOUND
 FLANN_LIBRARY_SHARED             /usr/lib/x86_64-linux-gnu/libflann_cpp.so
 FLANN_LIBRARY_STATIC             /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a
 FlyCapture2_INCLUDE_DIR          FlyCapture2_INCLUDE_DIR-NOTFOUND
 FlyCapture2_LIBRARY              FlyCapture2_LIBRARY-NOTFOUND
 FlyCaptureBridge_LIBRARY         FlyCaptureBridge_LIBRARY-NOTFOUND
 Freenect_INCLUDE_DIRS            Freenect_INCLUDE_DIRS-NOTFOUND
 Freenect_LIBRARY                 Freenect_LIBRARY-NOTFOUND
 Freenect_sync_LIBRARY            Freenect_sync_LIBRARY-NOTFOUND
 G2O_CLI_LIBRARY                  G2O_CLI_LIBRARY-NOTFOUND
 G2O_CLI_LIBRARY_DEBUG            G2O_CLI_LIBRARY_DEBUG-NOTFOUND
 G2O_CONFIG_FILE                  G2O_CONFIG_FILE-NOTFOUND
 G2O_CORE_LIBRARY                 G2O_CORE_LIBRARY-NOTFOUND
 G2O_CORE_LIBRARY_DEBUG           G2O_CORE_LIBRARY_DEBUG-NOTFOUND
 G2O_FACTORY_FILE                 G2O_FACTORY_FILE-NOTFOUND
 G2O_INCLUDE_DIR                  G2O_INCLUDE_DIR-NOTFOUND
 G2O_SOLVER_CHOLMOD               G2O_SOLVER_CHOLMOD-NOTFOUND
 G2O_SOLVER_CHOLMOD_DEBUG         G2O_SOLVER_CHOLMOD_DEBUG-NOTFOUND
 G2O_SOLVER_CSPARSE               G2O_SOLVER_CSPARSE-NOTFOUND
 G2O_SOLVER_CSPARSE_DEBUG         G2O_SOLVER_CSPARSE_DEBUG-NOTFOUND
 G2O_SOLVER_CSPARSE_EXTENSION     G2O_SOLVER_CSPARSE_EXTENSION-NOTFOUND
 G2O_SOLVER_CSPARSE_EXTENSION_D   G2O_SOLVER_CSPARSE_EXTENSION_DEBUG-NOTFOUND
 G2O_SOLVER_DENSE                 G2O_SOLVER_DENSE-NOTFOUND
 G2O_SOLVER_DENSE_DEBUG           G2O_SOLVER_DENSE_DEBUG-NOTFOUND
 G2O_SOLVER_EIGEN                 G2O_SOLVER_EIGEN-NOTFOUND
 G2O_SOLVER_EIGEN_DEBUG           G2O_SOLVER_EIGEN_DEBUG-NOTFOUND
 G2O_SOLVER_PCG                   G2O_SOLVER_PCG-NOTFOUND
 G2O_SOLVER_PCG_DEBUG             G2O_SOLVER_PCG_DEBUG-NOTFOUND
 G2O_SOLVER_SLAM2D_LINEAR         G2O_SOLVER_SLAM2D_LINEAR-NOTFOUND
 G2O_SOLVER_SLAM2D_LINEAR_DEBUG   G2O_SOLVER_SLAM2D_LINEAR_DEBUG-NOTFOUND
 G2O_SOLVER_STRUCTURE_ONLY        G2O_SOLVER_STRUCTURE_ONLY-NOTFOUND
 G2O_SOLVER_STRUCTURE_ONLY_DEBU   G2O_SOLVER_STRUCTURE_ONLY_DEBUG-NOTFOUND
 G2O_STUFF_LIBRARY                G2O_STUFF_LIBRARY-NOTFOUND
[... 106 lines cut ...]
 WITH_VINS                        OFF
 WITH_VISO2                       OFF
 WITH_XVSDK                       OFF
 WITH_ZED                         ON
 WITH_ZEDOC                       ON
 ZEDOC_INCLUDE_DIR                ZEDOC_INCLUDE_DIR-NOTFOUND
 ZEDOC_LIBRARY                    ZEDOC_LIBRARY-NOTFOUND           
 ZED_DIR                          ZED_DIR-NOTFOUND                     
 flann_DIR                        flann_DIR-NOTFOUND                         
 freenect2_DIR                    freenect2_DIR-NOTFOUND        
 g2o_DIR                          g2o_DIR-NOTFOUND
 grid_map_core_DIR                grid_map_core_DIR-NOTFOUND     
 k4a_DIR                          k4a_DIR-NOTFOUND               
 k4arecord_DIR                    k4arecord_DIR-NOTFOUND             
 libpointmatcher_DIR              libpointmatcher_DIR-NOTFOUND 
 libusb_INCLUDE_DIR               /usr/include                     
 mynteye_DIR                      mynteye_DIR-NOTFOUND
 netCDF_DIR                       /usr/lib/x86_64-linux-gnu/cmake/netCDF
 octomap_DIR                      octomap_DIR-NOTFOUND
 opengv_DIR                       opengv_DIR-NOTFOUND
 pnmutils_LIBRARY                 pnmutils_LIBRARY-NOTFOUND
 realsense2_DIR                   /home/.conan/data/realsense/2.54.2-pck-a/stable/package/lib/cmake/realsense2
 tmp                              tmp-NOTFOUND
 yaml-cpp_DIR                     yaml-cpp_DIR-NOTFOUND
```

bingogome, 2024-08-29:
I realized I did not install the G2O dependency. However, after installing G2O, I got some very jumpy results. Both translation and rotation became very unstable. Here is a recording. https://drive.google.com/file/d/1Z8zhv7aNKZkijUcokgZz_GstBfVU1Px4/view?usp=drive_link

matlabbe, 2024-09-05:
It should perform roughly the same on Windows and Ubuntu. If you search `OMP_WAIT_POLICY` on this [page](https://github.com/introlab/rtabmap/wiki/Installation), you will some issues depending on the OS config where there could be high artificial CPU usage. Other thing is how the dependencies are compiled between the OS. In your case, you seem to be using mostly system dependencies on Ubuntu, so it should be okay.

For your first videos, fast motion like that and pointing the camera to white wall are more likely to cause red screens, see description here https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens

For the jumps on the last post, it looks like related to IMU (the extrinsic between imu and camera may be wrong). Is it the same camera on both Windows and Ubuntu? Which kind of camera is it? You can try without IMU by detting IMU filtering to None here:
[image]

bingogome, 2024-09-05:
Thank you for getting back to me!
> For the jumps on the last post, it looks like related to IMU (the extrinsic between imu and camera may be wrong). Is it the same camera on both Windows and Ubuntu? Which kind of camera is it? You [...]

In all my tests, I use the same camera connected to the same PC with a dual boot. It is a RealSense 435i. I was also expecting the same behavior since the only difference I can think of is the dependency. Later, I also checked the settings options. I found one difference is the feature extraction. On Windows, I have GFTT+BRIEF, but it was grayed out on Ubuntu. On Ubuntu, it was GFTT+ORL.

Thank you for your suggestion to disable IMU. I will report back after the tests.