introlab/rtabmap #1333: Issue in Android Build

anuragSingh5exceptions, 2024-08-29:
HI team,
I love the work you have done 
I'm currently exploring the app and attempting to build an Android APK, but I'm having difficulty finding the relevant documentation on how to accomplish this. As an Android developer, I'm very interested in working on this project, but its structure is quite different from typical Android app code, which makes it challenging for me to proceed.

Could you please provide guidance or direct me to the documentation that explains how to build the app?

matlabbe, 2024-09-05:
Unfortunately, I was not an android developer, so the build is heavily cmake-based instead of using Android Studio.

The current Android pkg is built in CI following this script https://github.com/introlab/rtabmap/blob/master/docker/bionic/android/rtabmap_apiXX/rtabmap.bash

For convenience to have all dependencies ready for building, I suggest to open rtabmap in VSCode, then you will be asked if you want to open the dev container, click "Reopen in Container", then select "android":
[image]

Open a terminal and execute that script https://github.com/introlab/rtabmap/blob/master/docker/bionic/android/rtabmap_apiXX/rtabmap.bash from the root of rtabmap:
[image]

```
./docker/bionic/android/rtabmap_apiXX/rtabmap.bash /opt/android 30
```

[image]

To be able to install on android >=30, you need to sign the APK, you can follow instructions here: https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-arcore-apk

cheers,
Mathieu

anuragSingh5exceptions, 2024-09-06:
Hi Mathieu,

Thanks for your reply, I need one more favor. Could you please provide the source code for the Android-specific functionality? If it's available in your repository, that would be great. I'm eager to learn and would appreciate your help understanding how it works.

Thank you!

matlabbe, 2024-09-06:
See https://github.com/introlab/rtabmap/tree/master/app/android