introlab/rtabmap #1342: Resources for Using RTAB-Map in iOS App Development

AyushBharadwaj, 2024-09-14:
Hello,

I am an iOS developer interested in integrating the RTAB-Map open-source C++ library into an iOS application. Could you please provide any available documentation, guidance, or resources that would assist in using RTAB-Map in an iOS environment?

Specifically, I am looking for information on the following:
1. Setting up RTAB-Map for iOS.
2. Any available bindings or wrappers for using C++ libraries in Swift/Objective-C.

matlabbe, 2024-09-16:
We don't provide an interface library for swift/objective-c. There is an example of integration of RTAB-Map in iOS app here: https://github.com/introlab/rtabmap/tree/master/app/ios

To setup that project, see https://github.com/introlab/rtabmap/wiki/Installation#ios

In term of wrapper, while being very specific for the RTAB-Map iOS App, this [class](https://github.com/introlab/rtabmap/blob/master/app/ios/RTABMapApp/RTABMap.swift) is doing the bridge between Swift < - > objective-c < - > c++.

bahadirseyfi, 2024-11-05:
Hello, I am trying to integrate this application into an existing iOS project, but I’m not making any progress on how to do it. Do you have any tips or guidance on how to get started? Specifically, I’m looking for advice on how to bring these packages into an iOS project I’ve created.

Thank you again!

@matlabbe 
@AyushBharadwaj

DenisDaniel707, 2024-11-30:
> Hello, I am trying to integrate this application into an existing iOS project, but I’m not making any progress on how to do it. Do you have any tips or guidance on how to get started? Specifically, [...]

I also need this, @bahadirseyfi please update here if you find something

ioridev, 2024-12-11:
To integrate, check the RTAB-Map integration example in the iOS app, set ‘Search Path’ and ‘Frameworks,Libraries,and Embedded Content’ and change the C++ Language Dialect to gnu++14.