introlab/rtabmap #1343: Error in builfing for iOS (OpenCV)

th00tames1, 2024-09-16:
Hi, I encountered this build error after running install_deps.sh.
How can I fix this?

[image]

matlabbe, 2024-09-16:
The OpenCV libraries should have been built for iOS (not simulator) from here:
https://github.com/introlab/rtabmap/blob/69ac21f8116ca22d878e13941f4c67da2134e404/app/ios/RTABMapApp/install_deps.sh#L11
and then here:
https://github.com/introlab/rtabmap/blob/69ac21f8116ca22d878e13941f4c67da2134e404/app/ios/RTABMapApp/install_deps.sh#L135-L151

You may try to go under opencv build folder, and try rebuilding it with this explicit command:
```bash
cd rtabmap/app/ios/RTABMapApp/Libraries/opencv/build

cmake -G Xcode -DCMAKE_SYSTEM_NAME=iOS \
   -DCMAKE_OSX_ARCHITECTURES=arm64 \
   -DCMAKE_OSX_SYSROOT=iphoneos \
   -DBUILD_SHARED_LIBS=OFF \
   -DCMAKE_BUILD_TYPE=Release \
   -DCMAKE_OSX_DEPLOYMENT_TARGET=12.0 \
   -DCMAKE_INSTALL_PREFIX=../../ \
   -DOPENCV_EXTRA_MODULES_PATH=../../opencv_contrib/modules \
   -DBUILD_TESTS=OFF \
   -DBUILD_PERF_TESTS=OFF \
   -DWITH_CUDA=OFF \
   -DBUILD_opencv_apps=OFF \
   -DBUILD_opencv_xobjdetect=OFF \
   -DBUILD_opencv_stereo=OFF ..

cmake --build . --config Release

cmake --build . --config Release --target install
```
and ses if there are any errors.

th00tames1, 2024-09-17:
Solved. Thank you very much :)
Closing the issue.

closed 2024-09-17 by th00tames1