introlab/rtabmap #1345: Outdoor Settings

th00tames1, 2024-09-19:
Hi, I found the demo video link of Outdoor stereo SLAM with RTAB-Map: https://youtu.be/qpTS7kg9J3A
like this video, I tried to do the same work outside with my iPad. But it couldn't scan the shrub and vegetation normally.

[image]

Is there any suggested configuration for outdoor and plant scanning?
Thanks.

matlabbe, 2024-09-22:
You may try decreasing Depth Confidence in the settings. Here two comparisons between high (left) and medium (center and right):

[image]
[image]

th00tames1, 2024-09-25:
Thanks. Closing the issue

closed 2024-09-25 by th00tames1