introlab/rtabmap #1352: stereo camera example

nesquik011, 2024-10-08:
hello is there anyone has a C++ code as stereo example ? i want to do visalizatio and build a dense map with loop detection and icp etc 

thanks in advance

matlabbe, 2024-10-09:
Does the camera give you a side-by-side video stream in opencv? You could use this camera driver: https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/camera/CameraStereoVideo.h

You will need to calibrate the camera. If you have already the calibration, convert in the right format. Look at the yaml files in the example stereo data here: https://github.com/introlab/rtabmap/wiki/Stereo-mapping#process-a-directory-of-stereo-images

If the camera is not calibrated, you can calibrate from rtabmap main app. Go in Preferences->Source, select Stereo input then scroll down to select "Side-by-side video" driver, then click on calibrate. You can export a checkerboard from that screen and show it on a display/tv (then measure the size of the squares).

nesquik011, 2024-10-09:
@matlabbe thank you  , so the example i am working on will do as i want right ? 

[examples/RGBDMapping/main.cpp](https://github.com/introlab/rtabmap/blob/465ea85d91d8340ccc961f5997333e7e16c8258a/examples/RGBDMapping/main.cpp)

matlabbe, 2024-10-10:
Yeah, we don't really have other c++ examples :)

nesquik011, 2024-10-10:
@matlabbe i just built the lib and examples on windows 10 visual studio 2022 after suffering lol XD i will test the [CameraStereoVideo.h] with the example you shared and i will report back i hope nothing would break  

thanks a lot

nesquik011, 2024-10-11:
@matlabbe may you kindly check the example , i modified the RGBD mapping  , i keep getting warings 

`C:\Users\dream>C:\rtabmap\build\bin\rtabmap-rgbd_mapping.exe 0
Using single stereo camera (device=0)
[ WARN] (2024-10-11 22:05:32) CameraModel.cpp:378::rtabmap::CameraModel::load() Could not load calibration file "./00000001_left.yaml".
[ WARN] (2024-10-11 22:05:32) CameraModel.cpp:378::rtabmap::CameraModel::load() Could not load calibration file "./00000001_right.yaml".
[ WARN] (2024-10-11 22:05:32) CameraStereoVideo.cpp:184::rtabmap::CameraStereoVideo::init() Missing calibration files for camera "00000001" in "." folder, you should calibrate the camera!
[ERROR] (2024-10-11 22:05:43) CameraStereoVideo.cpp:230::rtabmap::CameraStereoVideo::init() Parameter "rectifyImages" is set, but no stereo model is loaded or valid.
[ERROR] (2024-10-11 22:05:43) main.cpp:71::main() Camera init failed!`

[https://pastebin.com/hRNVCUhv](url)

i have the yamls inside the exe folder as the following :  one for left and one for right , even pose , i am sure they are named 
00000001_left.yaml  and 00000001_right.yaml    

`%YAML:1.0
---
camera_name: m_left
image_width: 320
image_height: 240
camera_matrix:
  rows: 3
  cols: 3
  data: [ 1.0507460000000001e+03, 0., 9.6688520000000005e+02, 
           0., 1.0525362000000000e+03, 5.3855269999999996e+02, 
           0., 0., 1. ]
distortion_coefficients:
  rows: 1
  cols: 4
  data: [ 0., 0., 0., 0. ]
distortion_model: plumb_bob
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 9.9930298842773968e-01, -3.7330112769570531e-02, 0., 
           3.7330112769570531e-02, 9.9930298842773968e-01, 0., 
           0., 0., 1. ]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 1.0972377354488187e+03, 0., 9.5874497985839844e+02, 0., 
           0., 1.0972377354488187e+03, 5.5368847274780273e+02, 0., 
           0., 0., 1., 0. ]
local_transform:
  rows: 3
  cols: 4
  data: [ 0., 0., 1., 0., -1., 0., 0., 0., 0., -1., 0., 0. ]
essential_matrix:
  rows: 3
  cols: 3
  data: [ 0.95, 0.02, -0.03, 
           -0.02, 0.97, 0.01, 
           0.03, -0.01, 0.98 ]
fundamental_matrix:
  rows: 3
  cols: 3
  data: [ 0.0002, -0.0004, 0.005, 
           0.0004, 0.0001, -0.006, 
           -0.005, 0.006, 1.0 ]`

nesquik011, 2024-10-12:
i updated CameraStereoVideo.cpp with fourcc , but yet i cant manage to make it work 

`		if (src_ == CameraVideo::kUsbDevice)
		{
			if (stereoModel_.isValidForProjection())
			{
				if (capture_.isOpened())
				{
					// Set FOURCC to MJPG for USB camera
					capture_.set(CV_CAP_PROP_FOURCC, CV_FOURCC('M', 'J', 'P', 'G'));

					capture_.set(CV_CAP_PROP_FRAME_WIDTH, stereoModel_.left().imageWidth() * (capture2_.isOpened() ? 1 : 2));
					capture_.set(CV_CAP_PROP_FRAME_HEIGHT, stereoModel_.left().imageHeight());
					if (capture2_.isOpened())
					{
						capture2_.set(CV_CAP_PROP_FOURCC, CV_FOURCC('M', 'J', 'P', 'G'));
						capture2_.set(CV_CAP_PROP_FRAME_WIDTH, stereoModel_.right().imageWidth());
						capture2_.set(CV_CAP_PROP_FRAME_HEIGHT, stereoModel_.right().imageHeight());
					}
				}
			}
			else if (_width > 0 && _height > 0)
			{
				if (capture_.isOpened())
				{
					// Set FOURCC to MJPG for USB camera
					capture_.set(CV_CAP_PROP_FOURCC, CV_FOURCC('M', 'J', 'P', 'G'));

					capture_.set(CV_CAP_PROP_FRAME_WIDTH, _width * (capture2_.isOpened() ? 1 : 2));
					capture_.set(CV_CAP_PROP_FRAME_HEIGHT, _height);
					if (capture2_.isOpened())
					{
						capture2_.set(CV_CAP_PROP_FOURCC, CV_FOURCC('M', 'J', 'P', 'G'));
						capture2_.set(CV_CAP_PROP_FRAME_WIDTH, _width);
						capture2_.set(CV_CAP_PROP_FRAME_HEIGHT, _height);
					}
				}
			}
		}`