introlab/rtabmap #1353: elp camera kinda like a bug

nesquik011, 2024-10-08:
elp stereo cameras never work correctly unless fourcc('M', 'J', 'P', 'G')); is used so how could i do that in the GUI Windows ? 

and by any chance if anyone got code example for stereo camera please share it

matlabbe, 2024-10-09:
CameraStereoVideo driver uses opencv's VideoCapture to read from USB cameras (when single camera is used, it assumes it is a side-by-side stream). It opens the camera here:
https://github.com/introlab/rtabmap/blob/465ea85d91d8340ccc961f5997333e7e16c8258a/corelib/src/camera/CameraStereoVideo.cpp#L132

If opencv support `fourcc('M', 'J', 'P', 'G')` , you may check how to do it before it opens the camera and adjust the code above.

nesquik011, 2024-10-09:
@matlabbe   yes its side by side i will try to work with it , by the way they should add it to the GUI  so how to report that problem ? 

about opencv yes it support   `fourcc('M', 'J', 'P', 'G')`  i think from 4.4.0 all work the same way same API if i am not mistaken 

and thank you a lot for answering me <3

matlabbe, 2024-10-10:
> the way they should add it to the GUI so how to report that problem

do you mean adding `fourcc('M', 'J', 'P', 'G')` option under Preferences->Source and stereo video driver?

nesquik011, 2024-10-10:
i meant inside the GUI when i pick usb camera this option should be there , and the arrange is important `fourcc('M', 'J', 'P', 'G')`

for example 

cap width and hight 
fourcc('M', 'J', 'P', 'G')
cap fps 

this will never work , i test this with opencv 4.4.0 , 4.8.1 and some versions between them 

the correct way the fourcc('M', 'J', 'P', 'G') must be the last thing 

@matlabbe  and thanks so much for fast respond and being so kind with a new commers <3

matlabbe, 2024-10-11:
If you can confirm that you can get images doing what you suggest with [CameraStereoVideo](https://github.com/introlab/rtabmap/blob/master/corelib/src/camera/CameraStereoVideo.cpp) driver (even by hard coding it and running the main app), we could add indeed the option to GUI. 

cross-ref other posts: 
* https://github.com/introlab/rtabmap/issues/1352 
* http://official-rtab-map-forum.206.s1.nabble.com/stereo-example-tp10512p10524.html