introlab/rtabmap #1358: Coloring untextured surfaces with nearby Colors within the iOS App

electrum-bowie, 2024-10-19:
[image]
[image]

The vertex mesh assembling mode does this but unfortunately the textured parts are extremely low quality, and as you may know, not having textures on the model itself makes it incompatible with basic software.

electrum-bowie, 2024-10-19:
Also, if you add something like this, please keep the mesh enclosing feature ( hole filling functionality ) available, RTABMap is the only iOS software that provides actual complete 360 models, which even closes bigger voids in the capture that the LiDAR Scanner didn't reach !

It would be amazing to have a setting that can maximize the hole filling algorithm strength in the future as well, knowing that it can still leave some big gaps open. 

That combined with the requested feature in this post, additionally texturing meshes based on nearby colors, would finally enable full environmental scanning directly from the iPhone !

It would finally allow revisiting complete captures in VR, without having colorless or big voids in the models !

matlabbe, 2024-10-26:
Thanks for the suggestion. I'll check we can do that for next ios version release.

electrum-bowie, 2024-10-26:
Yup, using the PC to export a scan from a standalone device ( iOS ) to another standalone device ( Meta Quest VR ) isn't feasible.

electrum-bowie, 2024-12-06:
Does the new version include this feature ?

matlabbe, 2024-12-08:
Not yet.

DenisDaniel707, 2025-01-09:
This feature would be a game changer for sure! The app is already a powerhouse. Hats off to @matlabbe 🙏

matlabbe, 2025-01-13:
To add to the discussion, it seems that we can add vertex color in OBJ format, though it is not standard, so not sure about the support in software you may use. Ref: https://paulbourke.net/dataformats/obj/colour.html

If it works, that would be the easiest/fastest way to do it, as we should already know the closest color of each vertex in the mesh before texturing it.

matlabbe, 2025-03-15:
The new version with this new feature has been released to App Store.

You can enable this export mode by changing the following settings under Settings->Assembling:

[image]

This makes the mesh fully closed:
[image]

Note that there are two modes on how we attribute vertex colors:

[image]

By clicking on the Texture/Color blend visualization option, we better see the difference:

[image]

Depending on the software you are using, this is how the meshes may look like. To get the previous image with no seams, we have a custom shader to not apply vertex color when there is a texture.

When we export to OBJ, if we enabled vertex color, the color will be added to each vertex.

closed 2025-03-15 by matlabbe