introlab/rtabmap #1361: Add new camera driver

zhounao, 2024-10-21:
I use my own RGBD camera, What code should I add or modify so that my camera could Integrate your software?

matlabbe, 2024-10-26:
If you make a ROS driver, there would be no code to modify to make it work with rtabmap_ros. If you want to integrate in the main library / standalone app, look at the examples [here](https://github.com/introlab/rtabmap/tree/master/corelib/src/camera). Some drivers work as "on request" frame, other drivers are callback driven.

changewOw, 2025-01-16:
@matlabbe Hi, thank you for your great work.

I'm trying to figure out the input of the entire system.
`data = SensorData(rgb.imageRaw(), depth.depthRaw(), rgb.cameraModels(), rgb.id(), rgb.stamp());`
The above code is from `class CameraRGBDImages`

Here only the RGB camera parameters used, why there is no depth camera parameters, if there is no depth camera parameters, I don't understand how to calculate the point clouds.

Hence, does this mean that the input of rtabmap assumes that the user has aligned the depth to the RGB, so that only the RGB camera parameters should be used? And we can ignore depth camera parameters.

matlabbe, 2025-01-19:
> 1. Hence, does this mean that the input of rtabmap assumes that the user has aligned the depth to the RGB, so that only the RGB camera parameters should be used? And we can ignore depth camera parameters.

Exactly. The RGB calibration can then be used to project the point cloud, assuming the depth is already registered to RGB camera.

> 2. Is there a specific dataset example using this class class CameraRGBDImages as input?

I think we process the RGB-D SLAM Dataset with this class. https://github.com/introlab/rtabmap/blob/master/tools/RgbdDataset/main.cpp. We wrap the class into a SensorCaptureThread for the "postUpdate" function, but you can use the CameraRGBDImages class to capture the SensorData directly.

changewOw, 2025-01-21:
Thank you so much!

closed 2025-02-01 by matlabbe