introlab/rtabmap #1374: IOS: Possibility of Creating LAZ Files on iPhone?

zeitlerv, 2024-11-04:
Hello,

I'm working on a project that requires creating LAZ files directly on an iPhone, and I was wondering if you could provide any guidance on achieving this. I’ve encountered some challenges with existing libraries and would appreciate any suggestions or pointers you might have.

Here’s what I’ve tried so far:

LibLas: Unfortunately, LibLas only supports LAS up to version 1.3, which doesn’t meet my needs for LAZ file creation.

LasZip: I've managed to build it, but it only generates .dylib files, which aren’t easily compatible with iOS. Ideally, I would need a static library that could integrate smoothly with an iOS project.

PDAL: I also attempted to build a static library with PDAL, but I was unable to produce a version that works on iOS.

If you know of any solutions or could point me in the right direction for enabling LAZ file creation on iOS, I would be extremely grateful!

Thank you for your time and any help you can provide.

Best regards,
Viktor

matlabbe, 2024-11-05:
I tried building PDAL for iOS and got many issues with the dependencies. On main dependency is gdal, you may try this repo: https://github.com/BeinerChes/gdalios, though I could not make it work on my system.

LibLAS worked, though I didn't try building it with laszip dependency.
https://github.com/introlab/rtabmap/blob/75ac722996ca13e3c2443a50ef2b00e551d9e22c/app/ios/RTABMapApp/install_deps.sh#L225-L240

It [seems 2.0.1 is required](https://github.com/libLAS/libLAS/blob/62f3d57215ba1b074a84ff08a1539b69e69e38fb/CMakeLists.txt#L201-L211) with liblas: https://github.com/LASzip/LASzip/tree/2.0.1 . Maybe it can be compiled in a similar way than liblas above, then enable it as dependency to liblas to use it.

matlabbe, 2025-03-15:
The new version with this new feature has been released to App Store.

Go under Settings->Assembling..., scroll all way down and then you can change the format of point cloud exported:

[image]

closed 2025-03-15 by matlabbe

zeitlerv, 2025-03-17:
@matlabbe wow, well done!