introlab/rtabmap #1378: After building RTAB-Map with VINS-FUSION, rtabmap_viz show nothing when using roslaunch rtabmap_examples euroc_datasets.launch

WEILANZF, 2024-11-07:
Hello guys,I  have been stuck at this place for several days.

**envirment**

- ubuntu18.04
- ROS melodic

[image]

I have run this 

**roslaunch rtabmap_examples euroc_datasets.launch args:="Odom/Strategy 9 OdomVINS/ConfigPath ~/catkin_ws2/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml" MH_seq:=true raw_images_for_odom:=true

 rosbag play --clock MH_01_easy.bag**

```
ubuntu@ubuntu1:~/catkin_ws2$ roslaunch rtabmap_examples euroc_datasets.launch args:="Odom/Strategy 9 OdomVINS/ConfigPath ~/catkin_ws2/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml" MH_seq:=false raw_images_for_odom:=true
... logging to /home/ubuntu/.ros/log/bd2d71b8-9cbc-11ef-bbd8-2cfda1afc8cd/roslaunch-ubuntu1-9601.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu1:33691/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/
 * /rtabmap/stereo_odometry/

PARAMETERS
 * /imu_filter/publish_tf: False
 * /imu_filter/use_mag: False
 * /imu_filter/world_frame: enu
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/loc_thr: 0.0
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
[... 315 lines cut ...]
With XVisio SDK:         false
With libpointmatcher:     true
With CCCoreLib:          false
With Open3D:             false
With OctoMap:             true
With GridMap:            false
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:         true
With OpenVINS:           false

1. I have no idea what to do . I use the original launch file, change nothing.
2.  And I want to build RTAB-MAP with VINS-MONO, what should i do? VINS-FUSION can be easily link ,but i don't know how to link a similar one

Thank you all
Peace & Blessings

matlabbe, 2024-11-16:
For python error `AttributeError: 'module' object has no attribute 'FullLoader'`, maybe a python version issue. If the camera_info is not published, visual odometry will wait for topics.

On the turtlebot issue, there is this error:
`rtabmap: malloc.c:2401: sysmalloc: Assertion (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
`

See https://github.com/introlab/rtabmap/issues/872 and https://github.com/introlab/rtabmap_ros/issues/881 if it can help.

WEILANZF, 2024-11-20:
> For python error `AttributeError: 'module' object has no attribute 'FullLoader'`, maybe a python version issue. If the camera_info is not published, visual odometry will wait for topics.
> 
> On the turtlebot issue, there is this error: `rtabmap: malloc.c:2401: sysmalloc: Assertion (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed. `
> 
> See #872 and [introlab/rtabmap_ros#881](https://github.com/introlab/rtabmap_ros/issues/881) if it can help.

HI,matlabble,THANKS for help! I have change the system to ubuntu20.04, and follow the docker step by step.
**But I have some problem when build with ORBSLAM3v1.0.** 
environment
- ubuntu20.04
- ROS noetic
- vtk and pcl installed by sudo apt-get
-ORBSLAM3 PATH is right 

**cmake -DWITH_G2O=OFF -DWITH_GTSAM=ON -DWITH_CERES=ON -DWITH_VINS=OFF -DWITH_OPENVINS=OFF -DWITH_ORB_SLAM=ON ..**
CMAKE INFO
```
-- MOBILE_BUILD=OFF
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
CMake Warning (dev) at /usr/local/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (USB_10)
  does not match the name of the calling package (OpenNI).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindOpenNI.cmake:28 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:179 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:322 (find_openni)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:548 (find_external_library)
  CMakeLists.txt:240 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

CMake Warning (dev) at /usr/local/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
[... 601 lines cut ...]
[ WARN] [1732100178.763253937, 1403636578.947525179]: /rtabmap/stereo_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/stereo_odometry subscribed to (exact sync):
   /stereo_camera/left/image_rect \
   /stereo_camera/right/image_rect \
   /stereo_camera/left/camera_info \
   /stereo_camera/right/camera_info
[ WARN] (2024-11-20 18:56:19.669) Odometry.cpp:324::process() Updated initial pose from xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000 to xyz=0.000000,0.000000,0.000000 rpy=0.050821,0.290262,-3.090772 with IMU orientation
[ INFO] [1732100179.703274417, 1403636579.887496545]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.033540s, delay=0.123941s
[FATAL] (2024-11-20 18:56:19.786) OptimizerG2O.cpp:1906::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
[FATAL] [1732100179.787033761, 1403636579.968077726]: [FATAL] (2024-11-20 18:56:19.786) OptimizerG2O.cpp:1906::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2024-11-20 18:56:19.786) OptimizerG2O.cpp:1906::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
[ INFO] [1732100179.793604729, 1403636579.978157538]: rtabmap (1): Rate=1.00s, Limit=0.000s, Conversion=0.0071s, RTAB-Map=0.0655s, Maps update=0.0020s pub=0.0113s (local map=1, WM=1)
[... cut ...] header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap subscribed to (exact sync):
   /rtabmap/odom \
   /stereo_camera/left/image_rect \
   /stereo_camera/right/image_rect \
   /stereo_camera/left/camera_info \
   /stereo_camera/right/camera_info \
   /rtabmap/odom_info

```

WEILANZF, 2024-11-21:
> > For python error `AttributeError: 'module' object has no attribute 'FullLoader'`, maybe a python version issue. If the camera_info is not published, visual odometry will wait for topics.
> > On the turtlebot issue, there is this error: `rtabmap: malloc.c:2401: sysmalloc: Assertion (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed. `
> > See #872 and [introlab/rtabmap_ros#881](https://github.com/introlab/rtabmap_ros/issues/881) if it can help.
> 
> HI,matlabble,THANKS for help! I have change the system to ubuntu20.04, and follow the docker step by step. **But I have some problem when build with ORBSLAM3v1.0. Building success with some package warning. Error comes after first frame show in rtab_viz running the examples.(optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!)** environment
> 
> * ubuntu20.04
> * ROS noetic
> * vtk and pcl installed by sudo apt-get
>   -ORBSLAM3 PATH is right
> 
> **cmake -DWITH_G2O=OFF -DWITH_GTSAM=ON -DWITH_CERES=ON -DWITH_VINS=OFF -DWITH_OPENVINS=OFF -DWITH_ORB_SLAM=ON ..** CMAKE INFO
> 
> ```
> -- MOBILE_BUILD=OFF
> -- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
> CMake Warning (dev) at /usr/local/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
>   The package name passed to `find_package_handle_standard_args` (USB_10)
>   does not match the name of the calling package (OpenNI).  This can lead to
>   problems in calling code that expects `find_package` result variables
>   (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindOpenNI.cmake:28 (find_package_handle_standard_args)
>   /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:179 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:322 (find_openni)
>   /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:548 (find_external_library)
>   CMakeLists.txt:240 (FIND_PACKAGE)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
> -- The imported target "vtkParseOGLExt" references the file
>    "/usr/bin/vtkParseOGLExt-7.1"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
> but not all the files it references.
> 
> -- The imported target "vtkRenderingPythonTkWidgets" references the file
>    "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
> but not all the files it references.
> 
> -- The imported target "vtk" references the file
>    "/usr/bin/vtk"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
> but not all the files it references.
> 
> -- The imported target "pvtk" references the file
>    "/usr/bin/pvtk"
> but this file does not exist.  Possible reasons include:
> * The file was deleted, renamed, or moved to another location.
> * An install or uninstall procedure did not complete successfully.
> * The installation package was faulty and contained
>    "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
> but not all the files it references.
> 
> CMake Warning (dev) at /usr/local/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
>   The package name passed to `find_package_handle_standard_args` (libusb-1.0)
>   does not match the name of the calling package (PCL).  This can lead to
>   problems in calling code that expects `find_package` result variables
[... 748 lines cut ...]
[ERROR] [1732159733.725544761, 1403715289.209308101]: Make sure IMU is published faster than data rate! (last image stamp=1403715289.162143 and last imu stamp received=1403715289.157143)
[ERROR] [1732159733.777972242, 1403715289.259932724]: Make sure IMU is published faster than data rate! (last image stamp=1403715289.212143 and last imu stamp received=1403715289.207143)
[ERROR] [1732159733.828092231, 1403715289.310297071]: Make sure IMU is published faster than data rate! (last image stamp=1403715289.262143 and last imu stamp received=1403715289.257143)
[ERROR] [1732159733.876753241, 1403715289.350774069]: Make sure IMU is published faster than data rate! (last image stamp=1403715289.312143 and last imu stamp received=1403715289.307143)
[ INFO] [1732159733.993578971, 1403715289.473892396]: Odom: quality=197, std dev=0.013760m|0.010000rad, update time=0.052300s, delay=0.111749s
[ INFO] [1732159734.058657344, 1403715289.534316562]: Odom: quality=188, std dev=0.013760m|0.010000rad, update time=0.027979s, delay=0.072174s
[ INFO] [1732159734.103823665, 1403715289.584808594]: Odom: quality=214, std dev=0.013760m|0.010000rad, update time=0.025579s, delay=0.072666s
[... cut ...]
[ERROR] [1732159734.976103367, 1403715290.458243390]: Make sure IMU is published faster than data rate! (last image stamp=1403715290.412143 and last imu stamp received=1403715290.407143)
[ERROR] [1732159735.026225943, 1403715290.508653101]: Make sure IMU is published faster than data rate! (last image stamp=1403715290.462143 and last imu stamp received=1403715290.457143)
[ERROR] [1732159735.077544794, 1403715290.559189859]: Make sure IMU is published faster than data rate! (last image stamp=1403715290.512143 and last imu stamp received=1403715290.507143)
```

closed 2024-11-22 by WEILANZF