introlab/rtabmap #1379: Mapping and localization outdoors with respect to a car

shivamtrip, 2024-11-11:
Hi @matlabbe 

I am working on a project where I need to localize a drone (running PX4) outdoors wrt different parts of a car. I have an Oak-D Lite depth camera and I am thinking of using RTAB-Map for this, by first manually flying the drone to create a map of the car, and then autonomously commanding the drone to align with different parts of the car in localization mode. My origin (start of mapping) will likely be at a fixed distance from the front-left wheel. I am very inspired by your [rtabmap drone example](https://github.com/matlabbe/rtabmap_drone_example). 

I did some initial testing of this approach and have a few questions. Would really appreciate your suggestions.

1.  How can I improve the map of the car? Is there a way to remove the noise around the car, and use that edited map for localization? (I only want to map the car and not the environment, ground, etc. since the car might be parked in other places as well. For the current map I set `Grid/MinGroundHeight` to 0.5).
2. How can I improve localization wrt the car to achieve a cm level accuracy? (regardless of the environment in which the car was parked). 
3. Since I find that my visual odometry estimate drifts over time (using default rtabmap odom strategy), I am not sure if I can feed `rtabmap/odom` as a visual_pose estimate to PX4's EKF2 for localization. In that case, I am thinking to use tf from `map` to `base_link` frame as visual_pose_estimate and setting `RGBD/OptimizeFromGraphEnd` to true for dealing with jumps due to loop closure. Any advice here? 

[image]
[image]

Here is the [link](https://drive.google.com/file/d/17cubagXJttWtEJ-cQx_NGVC6HFnJoClu/view?usp=sharing) to the .db file for your reference (40 MB).

matlabbe, 2024-11-16:
Hi,

The object (car) you are scanning has reflective material and the sun can affect the camera itself depending of the point of view. The quality of the depth is not great (very noisy) or even nonexistent: 
[image]
[image]

> 1. How can I improve the map of the car? Is there a way to remove the noise around the car, and use that edited map for localization? (I only want to map the car and not the environment, ground, [...]

It depends at which level you want to remove noise. For example, if you just want to export the point cloud at the end, you may use these settings when exporting from rtabmap-databaseViewer:
[image]

Before:
[image]

After:
[image]

To remove better the ground, you would need to provide visual odometry aligned with gravity or feed IMU data to rtabmap node.

The other case is that you don't care how it looks, you only care about obstacle avoidance for the drone. If you set `Grid/MinGroundHeight=0.5`, the actual occupancy grid is empty, it is too high. 
[image]

Resetting to zero:
[image]

The occupancy grid like this already expose the car as obstacle cloud and the ground as ground (red=obstacle, green=ground). You could subscribe only to obstacle cloud topic. 
[image]

You can also generate an OctoMap for 3D trajectory planning around the car:
[image]

> 2. How can I improve localization wrt the car to achieve a cm level accuracy? (regardless of the environment in which the car was parked).

"regardeless of the environment", do you mean localizing against the same car in a different environment? That will be hard just visually. If the car had a matte finish to avoid reflections and the lighting would be constant, it wold be already better though not perfect. The car is also mostly textureless, so no features to localize on. We can see most features used are actually coming from the environment:
[image]
[image]

> 3. Since I find that my visual odometry estimate drifts over time (using default rtabmap odom strategy), I am not sure if I can feed rtabmap/odom as a visual_pose estimate to PX4's EKF2 for [...]

In rtabmap drone example, we are feeding back map->base_link to px4 controller like this: https://github.com/matlabbe/rtabmap_drone_example/blob/3055aea3a69da6c8acc0177c4c9c386578df38d5/src/offboard_node.cpp#L167-L182

We also don't use `RGBD/OptimizeFromGraphEnd=true` as we want the drone to move back to global position on loop closures.

cheers,
Mathieu

shivamtrip, 2024-11-21:
Thank you Mathieu, 

I really appreciate you taking the time to review my db file and providing such a detailed analysis! This really helps me understand the areas to focus on to improve my project results. And I didn't realize Rtabmap performs so many functions - what a powerful platform!

> The object (car) you are scanning has reflective material and the sun can affect the camera itself depending of the point of view. The quality of the depth is not great (very noisy) or even nonexistent:

About the mapping, I tested it on a car with a matte finish and indeed got a better map like you said.

> "regardeless of the environment", do you mean localizing against the same car in a different environment? That will be hard just visually. If the car had a matte finish to avoid reflections and the [...]

Got it! I will use the complete environment for localization in that case. 

----
I managed to integrate OpenVINS VIO odometry with RTABMap following [this discussion](https://github.com/introlab/rtabmap/issues/1181) and have some follow-up observations / questions for localization with my PX4 drone.

**1.** Even though standalone OpenVINS ROS2 version (without Rtabmap) worked on my PC (ROS2 Humble), I faced the below error when initially building it with rtabmap using the following command - ` colcon build --symlink-install --cmake-args -DWITH_OPENVINS=ON -DCMAKE_BUILD_TYPE=Release`

```
Starting >>> rtabmap 
Starting >>> rtabmap_msgs
Starting >>> rtabmap_python
/home/shivam/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
--- stderr: rtabmap_python                                                                                        
/home/shivam/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/shivam/.local/lib/python3.10/site-packages/setuptools/command/develop.py:41: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
---
Finished <<< rtabmap_python [0.93s]
Finished <<< rtabmap_msgs [1.37s]                                                      
[Processing: rtabmap]                              
--- stderr: rtabmap                                
CMake Deprecation Warning at examples/LidarMapping/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

In file included from /home/shivam/ov_ws/install/ov_msckf/include/core/VioManagerOptions.h:32,
                 from /home/shivam/ov_ws/install/ov_msckf/include/core/VioManager.h:34,
                 from /home/shivam/ros2_ws/src/rtabmap/corelib/src/odometry/OdometryOpenVINS.cpp:38:
/home/shivam/ov_ws/install/ov_msckf/include/state/StateOptions.h:25:10: fatal error: types/LandmarkRepresentation.h: No such file or directory
   25 | #include "types/LandmarkRepresentation.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1252: corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryOpenVINS.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:894: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2
---
Failed   <<< rtabmap [48.8s, exited with code 2]

Summary: 2 packages finished [48.9s]
  1 package failed: rtabmap
  2 packages had stderr output: rtabmap rtabmap_python
  11 packages not processed
```

I managed to bypass the above issue and successfully build by adding my OpenVINS package paths under `target_include_directories` in rtabmap corelib CMakeLists.txt. Is this a system path error on my PC, or did I miss a step?

---- 
**2.** What is the difference between building Rtabmap with OpenVINS support v/s just remapping the rtabmap odom topic to the OpenVINS odom topic, as in the above linked discussion?
 
Currently, I run OpenVINS separately using the following command -  `ros2 launch ov_msckf subscribe.launch.py config:=oak_d_px4`
Does building with rtabmap mean I can provide the config file location as a param in rtabmap and launch both together?

----
**3.** For my drone project, I am using OpenVINS for VIO, sending that to Rtabmap for loop closure / mapping, and then sending the real-time `map->base_link` tf as a visual odometry message to PX4. 

Just want to get your recommendation on whether this is a reasonable approach for good accuracy with outdoor camera-based localization on a drone, based on your experience. Would appreciate any suggestions, in case there are clear ways to improve this pipeline that I may not be aware of. (I am using OpenVINS since I observed slightly more drift using Rtabmap default odom approach (possibly because my URDF frame transforms were not super accurate - OpenVINS reads the Kalibr calibrated files directly).

matlabbe, 2024-11-24:
1. It could be a missing include directory from ov targets. Which openvins version are you using, we may need to update this if they now use cmake targets:
https://github.com/introlab/rtabmap/blob/5fff349623c5b1adcfdd47e50f6bf75af310f318/corelib/src/CMakeLists.txt#L746-L755

2. Inside rtabmap, you may not need any other config files, as calibration data and extrinsics will be taken from TF and the camera_info messages directly. If the camera is not the base_link, I don't know if openvins can handle that case, but rtabmap does (e.g., the camera could be on head of the robot and still estimate correctly accordingly to the base frame even if the head rotates). That may be more convenient in some cases. However, beside that, if you need to set more custom parameters to openvins, you may be better using externally like you did, and also easier to debug openvins part. Just make sure that loop closure detection is disabled on their side.

3. That sounds reasonable to me!

shivamtrip, 2024-11-24:
> 1. It could be a missing include directory from ov targets. Which openvins version are you using, we may need to update this if they now use cmake targets:

I cloned and installed their latest version (master branch) from [here](https://github.com/rpng/open_vins). Let me know if you'd like me to test any changes you make.

That's all the questions I had for now. Thank you so much for your help! :)

closed 2024-11-24 by shivamtrip